luxonis/oak-hardware #15: Support ArduCam 155-degree FishEye OV9282 Camera

Luxonis-Brandon, 2020-08-05:
### Start with the `why`:

As it stands now DepthAI is optimized for real-time spatial AI, so where objects are in physical space and attributes about them.  So for this, narrower field of view is desirable.

But for applications where it is desirable to self-orient and self-locate (i.e. simultaneous location and mapping - SLAM), wider field of view is of interest.  So having support for a wide field of view OV9282 camera module is of interest.

And it just so happened that Arducam reached out offering cooperation, and has a module that is ideal in terms of FOV:

- 155 degrees HFOV

[image]

### Move to the `how`:

We are working with ArduCam to integrate this OV9282 fisheye version in with DepthAI.

### The `what`:

Support ArduCam 155-degree FishEye OV9282 Camera

george-hawkins, 2020-08-05:
Is this a module that ArduCam are working on bringing to market?

I couldn't find such an OV9282 based module in their current lineup - the nearest I could find was this US$30 [170&deg; OV9281 based camera](https://www.arducam.com/product/arducam-ov9281-mipi-1mp-monochrome-global-shutter-camera-module-raspberry-pi/).

Luxonis-Brandon, 2020-08-05:
Thanks @george-hawkins - great question - and thanks for catching my error(s).  So I had misinterpreted it as a OV9282 whereas it is actually OV9281, and added the link you shared to the Github issue above.

So it is 166 degree diagonal FOV and 155 horizontal FOV - so that's the discrepancy in FOV we were respectively seeing.

ArduCAM, 2020-08-05:
@george-hawkins I gave Brandon the wrong info, it should be 166 degrees in horizontal. Extremely wide, almost fisheye and without black corners.

ArduCAM, 2020-08-05:
Maybe I remember incorrectly, 166 degrees should be diagonal, but extremely wide and no black corners.

slitcch, 2021-01-15:
So, could I literally just buy the camera from Arducam and stick it on my OAK-D? Or are there still some outstanding issues that need to be solved?

Luxonis-Brandon, 2021-01-15:
These are firmware compatible as of now (which is the big hurdle) but the connector pinout for the module is different.  So a variant of the OAK-D carrier board would need to be use.  Also, for calibration accuracy, the camera internal to OAK-D are actually adhered to the board, so they are not swappable.

I am not sure if ArduCam has already made this variant.  @ArduCAM ?

ArduCAM, 2021-01-16:
Sorry, we didn't build any drop-in replacement variant modules for OAK-D yet. 
@slitcch you might have to use BW1098FFC board to connect our OV9281 camera module with an adapter board for larger FOV. 
https://www.uctronics.com/arducam-ov9281-mipi-camera-module-1mp-global-shutter-mono-camera-130.html

rdthompson2020, 2021-01-21:
> Sorry, we didn't build any drop-in replacement variant modules for OAK-D yet.
> @slitcch you might have to use BW1098FFC board to connect our OV9281 camera module with an adapter board for larger FOV.
> https://www.uctronics.com/arducam-ov9281-mipi-camera-module-1mp-global-shutter-mono-camera-130.html

This Q about a Wider FOV FishEye Camera hits the nail on the head for what's really needed in OAK-D
Purchased the OAK-D and waiting on it but most applications my group is interested in are FPV tracking on a Drone and aligning with footage camera. I understand there may be mapping differences for OpenCV models but if there's a way, I think this question for OAK-1 in general and later models is going to pop up a lot.

Luxonis-Brandon, 2021-01-23:
Thanks.  So after CNY we will likely have prototypes with these modules available.  We plan to sell the ArduCam modules directly as well to make this easier for customers.

Thanks,
Brandon

Luxonis-Brandon, 2021-01-27:
As an update @ArduCAM made a little adapter so that these wide-angle can be used directly on the BW1098OBC/BW1098OAK.  

[image]
[image]
[image]
[image]

Luxonis-Brandon, 2021-01-28:
Got these working on DepthAI today with @ArduCAM 
[image]

Luxonis-Brandon, 2021-01-28:
[image]
[image]

Luxonis-Brandon, 2021-02-01:
@ArduCAM just shared that they made a fully-pin-compatible 2-lane version:
[image]

Luxonis-Brandon, 2021-02-17:
Here's the comparison of the first prototype vs. the final modules:
[image]

Luxonis-Brandon, 2021-03-03:
This is ArduCam part number B0297-C.

closed 2021-05-24 by Luxonis-Brandon