luxonis/oak-hardware #20: Dual BELICE-SD Dot Projector + IR-capable Global Shutter Grayscale

Luxonis-Brandon, 2020-09-16:
### Start with the `why`:

For high-granularity depth and for support of low-visual-interest objects, having a dot projector to augment these surfaces with visual interest is extremely beneficial.

### Move to the `how`:

Leverage IR-capable camera modules (either visible-light + IR or IR-bandpass) and a [BELICE-SD](https://ams.com/belice) dot projector pair (and timing/strobe circuitry for more efficient power use) to make an active-illumination variant of DepthAI.

### Move to the `what`:

Support active illumination on DepthAI.

eric-schleicher, 2020-11-17:
this is great.  how much are the dot projectors?

Luxonis-Brandon, 2020-11-17:
I don't -think- we can share.  Or at least we haven't asked.  But here they are: 
[image]

We're behind on board design help right now.  So we haven't made a reference with these yet, but we're hiring so we should be able to early next year.  

And then need to go through laser safety certification.

Luxonis-Brandon, 2021-01-31:
The laser safety certification is looking to be a blocker.  We engaged with a consultant on this, but it kind of stopped there.  So we are trying to re-kick it off.

Luxonis-Brandon, 2021-11-12:
This is now implemented in https://github.com/luxonis/depthai-hardware/issues/114

closed 2021-11-12 by Luxonis-Brandon