luxonis/oak-hardware #23: Step file created for BW1098OBC

ZurMaD, 2020-10-22:
Hello there, I build a simple files for a version of BW1098OBC for people who just want a model with real dimensions and no details, maybe the Depth AI can add to documentation to support community

[image]

Dimensions according to https://github.com/luxonis/depthai-hardware/tree/master/BW1098OBC_DepthAI_USB3C

Public link of the files https://grabcad.com/library/bw1098obc-for-oak-d-stereo-camera-1

Also, for BW1093

[image]

Public link of the files https://grabcad.com/library/bw1093-for-oak-1-color-camera-opencv-1

Luxonis-Brandon, 2020-10-23:
Thanks @ZurMaD !  So one note on OAK-1 is that the fins actually do go the opposite direction:
[image]

ZurMaD, 2020-10-23:
Thanks for the feedback, I tried to fix it. It's not exactly the same heatsink fin, but I think it's ok. Next week, I'll upload new versions of two models direct to GrabCad.

[image]

Luxonis-Brandon, 2020-10-23:
Thanks @ZurMaD !

closed 2021-01-12 by ZurMaD