luxonis/oak-hardware #24: How Depthai hardware is different than intel real sense

yashhema, 2020-10-26:
Hello,
I got to know of luxonis products yesterday while searching for depth camera, I saw real sense then I saw luxonis.

Can you please elaborate - how these products are different (intel realsense d4xx series + movidius)

Luxonis-Brandon, 2020-10-26:
Hi @yashhema ,

Great question.  So we actually have a forum post that describes the differences.  In short you nailed it though, DepthAI is effectively an Intel RealSense d4xxx series + Myriad X in a tightly-integrated product.

Here's more details though:

1. Contrasting DepthAI with existing depth cameras: [https://discuss.luxonis.com/d/48-summary-of-stereo-mapping-cameras](https://discuss.luxonis.com/d/48-summary-of-stereo-mapping-cameras)
2. How DepthAI allows integration into actual products (and modalities of doing so): [https://docs.luxonis.com/faq/#how-do-i-integrate-depthai-into-our-product](https://docs.luxonis.com/faq/#how-do-i-integrate-depthai-into-our-product)

Does that help?  

Thanks,
Brandon

yashhema, 2020-10-27:
Hello,
Thanks for the answer. What about the distance range , like D435 can calculate depth till distance of 10 m , and D455 till almost 20 meters.
And does the depth measurement , changes between day and night  (ie it reduces in night as compared to day)

Luxonis-Brandon, 2020-11-02:
Sorry I was delayed on this one.  I thought I had answered but apparently not.  So we haven't done a whole lot of characterization of depth distance yet.  But we support variable stereo baseline which makes it so long-range is feasible.  
[https://shop.luxonis.com/collections/all/products/depthai-usb3-edition](https://shop.luxonis.com/collections/all/products/depthai-usb3-edition)

We are currently passive entirely.  But ArduCam is making camera modules that allow active illumination like the BELICE-SD ([here](https://github.com/luxonis/depthai-hardware/issues/20)) and overall IR flood illumination ([here](https://github.com/luxonis/depthai-hardware/issues/22))

So this would allow zero-light performance.  For longer-distance more focused optics (implementable with M12, see [here](https://github.com/luxonis/depthai-hardware/issues/17)) and illumination may be of interest.  Or just higher-brightness IR illlumination.

Thoughts?

Thanks,
Brandon

Luxonis-Brandon, 2020-11-13:
Going to close this @yashhema for now but feel free to re-open it.  But to leave with an image, we just got subpixel (https://github.com/luxonis/depthai/issues/163) running.  Here's a view without any filtering.  (Note that distanced greater than 2 meters were thrown away by the Open3D visualization, that's why you don't see behind me in the point cloud... it's just a setting in Open3D visualization I forgot to change.)

[image]

We'll be adding Bilateral filtering onboard as well (i.e. without any need for the host to do anything).

closed 2020-11-13 by Luxonis-Brandon