luxonis/oak-hardware #28: Raspberry Pi Compute Module 4

Luxonis-Brandon, 2020-12-12:
We're leveraging the BW1097 to make a variant with the Pi Compute Module 4 instead of the Pi Compute Module 3.

[image]
[image]

neixlo, 2021-01-13:
Im highly interested in this build due to the CM4!

The perfect match for me would be the **BW1093+CM4+PoE**+(maybe global shutter rgb instead the imx roling shutter).
You aren't planning something like that as well, or do you?

I might be able to use this BW1097/CM4 design as reference, thanks!

neixlo, 2021-01-13:
Just stumbled across this [BW2098POE ](https://github.com/luxonis/depthai-hardware/tree/master/BW2098POE_PoE_Board). It seems that I look for something like the BW2098POE, except of the CM4 which I'll use as host.

Luxonis-Brandon, 2021-01-13:
Hi neixlo,

Thanks for the comments.  So as an update the boards are now back, but not yet fully assembled:
[image]

And tagging @Luxonis-David for doing a PR around this design so you can have access.  I think it's a great idea to use this as a reference.  

We do have a global-shutter color plan, but it's the OV9782, so it's 1280x800 resolution global shutter color.  We have it initially running in https://github.com/luxonis/depthai-hardware/issues/21 , but we haven't done the debayering to actually get the color out yet.

Is it high enough resolution for your needs?

Thanks,
Brandon

neixlo, 2021-01-14:
Hi @Luxonis-Brandon ,
these boards are looking really good!

Not sure if this Resolution will be enough, I might need 1920x1080. Maybe even more, but this is not sure yet.
Anyway, at the moment I try to get a sense what's coming and what's possible to have a rough plan for future.

Right now I dig into the OAK1 while using an external host.

Thanks a lot!

Luxonis-Brandon, 2021-01-14:
Sounds good, thanks.  The AR0234 is probably good for what you are thinking for global shutter color than.  We haven't prototyped with it yet, but we want to eventually integrate it.

Luxonis-Brandon, 2021-03-02:
I should have closed this a while ago.  Just noticed.  

This is now fully implemented and vetted:
https://github.com/luxonis/depthai-hardware/tree/master/DM1097_DepthAI_Compute_Module_4

And available for purchase here:
https://shop.luxonis.com/collections/all/products/depthai-rpi-compute-module-4-edition

closed 2021-03-02 by Luxonis-Brandon