luxonis/oak-hardware #30: Support IMX283

Luxonis-Brandon, 2020-12-23:
### Start with the `why`:

For low-light color and long-range high-quality digital zoom applications, it is beneficial to have a large image sensor with large pixels.  The IMX283 provides both, with 20MP and a 15.86 mm optical diagonal sensor size (quite large) with 2.4um square pixels.

### Move to the `how`:

ArduCam is helping us to make IMX283 sensor boards that work with DepthAI.  We will then be writing a driver for this module, and see how much tuning is required.

### Move to the `what`:

Support the IMX283 on DepthAI.

Luxonis-Brandon, 2021-01-27:
[image]

Luxonis-Brandon, 2021-02-17:
We got this working in firmware today:

[image]
[image]

yq3w, 2023-09-07:
> [image]

Hello, where can I find the driver for IMx283?

Erol444, 2023-09-07:
Hi @yq3w ,
IMX283 sensor is currently supported on the `imx283` branch on the depthai-python repository. You can checkout to that branch and call `examples/install_requriements.py` to install correct version of depthai that supports IMX283.

alex-luxonis, 2023-10-31:
Was just merged to `develop`.