luxonis/oak-hardware #36: BW1098FFC with Onboard IMU and ArduCam FFC connections

Luxonis-Brandon, 2021-01-19:
### Start with the `why`:

The ArduCam OV9281 and IMX477 and other modules that offer M12, wide FOV, C-mount, etc. are extremely useful for making depthai applicable for varieties of use-cases.  So it would be beneficial to have direct support for them one the BW1098FFC without [any adapters](https://github.com/luxonis/depthai-hardware/tree/master/BW0253_R0M0E0_RPIHQ_ADAPTER).

Also, in many applications, the IMU present on the OAK-D, BW1092, etc. would be beneficial, so when modifying this board, it would make sense to add the IMU.

### Move to the `how`:

Pull the IMU from the BW1098OAK and change the FFC adapter connectors to be what ArduCam uses on their FFC cameras.

### Move to the `what`:

Support the ArduCam camera modules directly without an adapter.

Luxonis-Brandon, 2021-01-20:
So ArduCam actually had independently proposed making this yesterday.  So they are already building it.  :-)

Luxonis-Brian, 2021-02-02:
Updated this issue to add more detail to the `what` section

Luxonis-Brandon, 2021-02-02:
Thanks!  And I think ArduCam submitted designs to us with these changes already... did they?

Luxonis-Brandon, 2021-03-11:
This has now been implemented and ArduCam just built a prototype board and verified that it all works!

[image]

Luxonis-Brandon, 2021-03-19:
And the PR for this is here: https://github.com/luxonis/depthai-hardware/pull/41

Luxonis-David, 2021-03-22:
Pull request #41 is for old interface with more robust connectors, PR for 26-pin interface with IMU support is here #50.

Luxonis-Brandon, 2021-03-27:
The fully updated versions just finished being made.  DM1090FFC is the new name, changed from 1098FFC since these have different connectors and we will continue to sell the old ones for those than need them on existing installations.
[image]
[image]

closed 2021-04-01 by Luxonis-Brandon

Luxonis-David, 2021-04-12:
Resultant files for this issue can be found on above linked pull-request or on the link below: 
[DM1090FFC_DepthAI_USB3](https://github.com/luxonis/depthai-hardware/tree/master/DM1090FFC_DepthAI_USB3)