luxonis/oak-hardware #44: BW1099EMB Heatsink / 3D STEP

AndrewKeene, 2021-02-19:
Hi DepthAI team, 

I see the 3D models of the carrier boards have the heatsink model integrated, but I'm not seeing the heatsink in the standalone SoM models.  Is there a 3D STEP available for the BW1099EMB with the heatsink mounted, and if not is there a STEP just for the heatsink available?  Thank you!

Luxonis-Brandon, 2021-02-19:
Thanks @AndrewKeene ,

Good point.  We should add a version of each that includes the heatsink, for each SOM [here](https://github.com/luxonis/depthai-hardware/tree/master/SoMs).  @Luxonis-David - could you please do so when you have a second?

And in the meantime I'll see if I can find the STEP file for the heatsink by itself.

Thanks again,
Brandon

Luxonis-Brandon, 2021-02-19:
Here is the step file for the heatsink @AndrewKeene :  

[https://drive.google.com/file/d/16t6zTeiB6AHo2_8wcAw2twctY_KXrG8R/view?usp=sharing](https://drive.google.com/file/d/16t6zTeiB6AHo2_8wcAw2twctY_KXrG8R/view?usp=sharing)

[image]

We'll get it added to the Git directories and with the SOM together next week as well.

Thoughts?

Thanks,
Brandon

Luxonis-David, 2021-02-19:
I have added step files for each SOM [here](https://github.com/luxonis/depthai-hardware/tree/master/SoMs).
Also updated readme.

AndrewKeene, 2021-02-19:
Thanks guys!

closed 2021-02-19 by AndrewKeene