luxonis/oak-hardware #45: DM1097 CM4 based module doesn't boot without HDMI connected to a monitor

mslavescu, 2021-02-28:
This is about this model:
https://github.com/luxonis/depthai-hardware/tree/master/DM1097_DepthAI_Compute_Module_4

I set this option, but still no boot when HDMI is not plugged:

https://www.raspberrypi.org/documentation/configuration/config-txt/video.md
hdmi_force_hotplug=1

Are there other settings required?

Is this a know issues or is just my board problem?

mslavescu, 2021-02-28:
I found workaround, but would like to have a way to boot without any HDMI connected.

Using this adapter (or any other HDMI to USB adapter) I was able to boot and see the video on my phone using Android FPVViewer app:
HDMI Video Capture Card HD 1080P
https://www.amazon.ca/dp/B0882XY5Q4
Easycap & UVC Player(FPViewer)
https://play.google.com/store/apps/details?id=labsp.android.viewer

See attached a demo video.
Once the RPI CM4 boots I can disconnect it and it will continue to work until next reboot.

https://user-images.githubusercontent.com/461669/109433891-a69eeb00-79e0-11eb-8357-93c5a58a29f0.mp4

Luxonis-Brandon, 2021-03-02:
Thanks @mslavescu .  This is so strange it won't boot without HDMI.  The Pi CM4 is still relatively new to us.  So I guess it's not a surprise to some degree.  We will look into how/why booting on CM4 doesn't work w/out HDMI.

Luxonis-Brandon, 2021-03-02:
I think I found what you had found @mslavescu: 

https://www.raspberrypi.org/forums/viewtopic.php?t=253312

> Adding hdmi_force_hotplug=1 to /boot/config.txt seems to have solved the problem. The Pi4 is running headless.

Later on it mentions:

> Use sudo raspi-config to set a forced screen resolution.

> The reason is that by default (RPi4), if no screen is connected at boot then a display device is not created. Without a display device the GUI desktop does not start so any program that requires GUI will not start. Other RPi models did not have that issue because they would fall back to composite mode if no HDMI was connected. The RPi4 has composite mode disabled by default so no display device is created.
> Setting a resolution mode with raspi-config or the GUI config tool or manual edits to config.txt as mentioned above, will force a display device on boot even without the HDMI connected as long as you don't choose the monitor default setting.

So it sounds like this is it @mslavescu !  So you could alternatively just enable composite mode.  It is disabled by default on the Pi 4 series.

Thoughts?

Thanks,
Brandon

mslavescu, 2021-03-03:
Thanks @Luxonis-Brandon!
That did the trick, I set a custom resolution in raspi-config and it booted just fine after that.

closed 2021-03-03 by mslavescu

Luxonis-Brandon, 2021-03-03:
Sweet thanks!

Luxonis-Brandon, 2021-05-16:
This solution above seems to no longer work for the Pi 4 CM booting without HDMI attached, as noted/debugged [here](https://discuss.luxonis.com/d/204-communication-exception/6).

To get the Pi 4 CM to boot up completely without HDMI attached, adding this to /boot/config.txt did the trick:

`enable_tvout=1`

Please note though that for me this actually completely prevented HDMI from working at all. But I think for the use-case here, that's fine.