luxonis/oak-hardware #47: Battery options/recommandations for all models

mslavescu, 2021-02-28:
I tried different 5V USB batteries (LiIon based) that output 2A, 2.4A and 3A and seems to be fairly stable on DM1097 CM4 models, the most power hungry.
I will run some endurance tests and provide more details once I mount them on my robots.

Let's collect here what are your recommendations to power by battery the DepthAI modules.

Also please provide the min/max voltage that can be used and what temperatures we should expect when the voltage is on lower end, for each model. 

I just noticed that the Myriad heat-sink temp is about 57 degree Celsius, while running the demo (on API version 1.0) and using a DeWALT 20V 2Ah battery using a 18V convertor that has a USB port that outputs 5V at 2A max.

Here is the adapter, I assume the voltage is a bit lower than 5 under use (need to measure it):
https://www.amazon.ca/TENMOER-DCA1820-Battery-Adapter-Compatible/dp/B08HMVYYG5

mslavescu, 2021-02-28:
On this battery I ran for about 36 min (load and no load) and lost 1 bar, I assume it can work about up to 1.5 hours on it:
https://www.amazon.ca/DEWALT-DCB203-2-Compact-Li-Ion-Battery/dp/B00BYKVMSY
with this adapter:
https://www.amazon.ca/TENMOER-DCA1820-Battery-Adapter-Compatible/dp/B08HMVYYG5

Luxonis-Brandon, 2021-03-01:
So those batteries are great but actually intended for extremely-high continuous amperage (let’s say 100 Amps, but I don’t know the actual number for that battery, it might say on it or on the website), so they’re higher cost/amp-hour and usually lower energy density of the type of batteries in say USB battery banks, which only put out let’s say 3 Amps at 3V to 3.7V.

We can source a 10Ah 3.7V battery for $5 at Q=1 for reference.  We could probably sell these on the store too... just not sure about the special shipping required.

Luxonis-Brandon, 2021-03-02:
Here's the battery (being charged):
[image]

closed 2021-07-09 by Luxonis-Brandon