luxonis/oak-hardware #51: BW1099EMB NOR Flash Communications

AndrewKeene, 2021-03-29:
I'm having issues communicating with the NOR flash on the BW099EMB through the DepthAI Embedded Development Kit.

In the BW1099EMB datasheet, it states there are 1K Pull-Up resistors on the SPI0_SIO2 and SPI0_SIO3 signals, which are hardwired to the NOR flash ~WP and ~HOLD signals.  The dev kit doesn't have the ~WP and ~HOLD signals routed to the Molex connectors where I can control them externally, so I've probed the MT25Q flash physically on pin 7 and 3, and both were 0V. 

I'm assuming the issue is the ~HOLD signal being asserted which is preventing proper SPI communications.  Is there a way to verify the pull-up resistors on the SoM, or maybe something is pulling ~WP and ~HOLD down?  I've performed combinations of holding the DepthAI in reset and not in reset, and both cases ~HOLD was asserted to ground. 

Thanks guys.

Luxonis-Brian, 2021-03-29:
So you can actually access these on the baseboard (I'm assuming you have BW1098EMB, but all the EMB baseboards are very similar). It's true that these aren't broken out through the Molex connector interface, but there are test points which will be easier to access than the pins on the NOR flash. 
[image]

Also, on the 1099EMB, it's resistors R70 and R71 that pull up those lines to 1.8V. I think maybe you have a 1099 (non-EMB) if you were able to probe on the pins? That's the one with the 8MB flash and SOP-style package. They should be the same in this regard, though. I agree, it's odd that those lines are held low even during RST.....

AndrewKeene, 2021-03-29:
Hi Brian,

Yes, it looks like I was probing the EEPROM (or something else) assuming it was the NOR Flash. I do have the BW1099EMB SoM.  I see now after further inspection it's a WinBond part and the Micron is the flat package.  OK, disregard the ~HOLD and WP issue, as it was completely the wrong part I was probing.  

I'll keep looking into this, but maybe there is something I'm not understanding with the PU situation on ~HOLD and ~WP.  I see in the BW1092 and BW1098 I/O table it states "no" for 1099 PU/PD column on SPI0, but the datasheet says there is (screenshots below).

Should I move forward with the understanding that SPI0_SIO2 and SPI0_SIO3 should state "yes" for PU based on the datasheet, or is that column representing something else?

Thanks for help!

BW1092 Table ('no' for PU):
[image]

Datasheet:
[image]

BW1098EMB Table ('none' for PU/PD):
[image]

Luxonis-Brian, 2021-03-29:
Oof - I think you found an error in the schematic GPIO table. There are definitely 1k pull ups for SPI0_SIO2 and 3. The datasheet is correct. We'll have to correct that schematic page and re-upload!

Sorry for the error! I hope you didn't have too much time wasted here!

AndrewKeene, 2021-03-29:
Ok no problem, I'm glad I now have a better understanding of the BW1099EMB SoM PCB.  I'm essentially prepping a test plan for our custom PCB and the NOR Flash is the critical item.

Now that I know there are definitely pull-ups on ~HOLD and ~WP, that means there is something I'm still not doing correctly to be able to read the JEDEC ID through the SPI0 exposed pins on the DepthAI Dev Kit.  Closing the ticket to continue debugging, but can you confirm that I should be able to read and write the Flash from an MCU or Linux through those pins when the MyriadX is in reset?

Thanks Brian!

closed 2021-03-29 by AndrewKeene