luxonis/oak-hardware #53: LUX-D-POE-RCM4

Luxonis-David, 2021-04-13:
### Start with the `why`:

There is a lot of interest in the community to have a decent Linux host integrated with DepthAI.  Having all enclosed (possibly IP rated) and PoE powered would give a lot of options to use the device for final integration. The intention is to use PoE as a power source for the device.

### Move to the `how`:

The same PoE design can be used, same as for [LDC-D-POE](https://github.com/luxonis/depthai-hardware/tree/master/SJ2088POE_PoE_Board) and [LDC-1-POE](https://github.com/luxonis/depthai-hardware/tree/master/SJ2096POE_R0M0E0) meaning by default only Ethernet port would be accessible when the device is in the enclosure. Connectors like micro-HDMI, micro-USB should be also added for development and debugging purposes.  

Due to having a full-blown Linux alongside the depthAI it would be good to have SoM connected over some fast speed interface so we were thinking of using PCIe to USB3 bridge as it is used on RPi 4B+ and connect depthAI SoM over USB3 interface.

Later we came to the idea that it would be the best to have CM4 and SoM connected directly over PCIe and thus making the most of it and reducing the BOM. At the moment we could not find relevant info about if RPi CM4 supports needed plug-and-play PCIe so we will have to test this out with the first prototypes. We could also do some workarounds in FW if needed to get it working, only if PnP is not supported on CM4. 

As a workaround (if PCIe fails to be a good idea) USB2 interface will still connect CM4 with SoM as on previous versions of 1097 devices. USB connection is also great to have for flashing NOR on DepthAI SoM if needed for recovery mode. 

As for some use cases, it is great having the IMU/magnetometer on the baseboard as it brings extra functionalities to the device. 
The device should use a bare minimum of components needed to implement all of the above features. 

### The `what`:

Make a design that can be easily placed in an enclosure try to make it as small as possible. 
Add PoE functionality as a power source, remove all unnecessary connectors/interfaces like CSI, DSI, barrel jack power, audio,... Use smaller connectors where possible. Add IMU and other features which are possible to fit onboard to make first prototypes better for evaluation and testing of as many features as possible. Some connectors/circuits can be populated only on a few devices for easier bring up and later removed in production.

Luxonis-David, 2021-04-13:
Here is what we're thinking so far: 
[image]
[image]
[image]

Luxonis-David, 2021-05-05:
At latest we decided to go with the proven design solution which would be using PCIe to USB3 bridge as it is used on RPi 4B+.
We have connected depthAI SoM and CM4 over USB3 interface to bring more speed to the device and being capable transferring more data at the same time. 
There is also a PR #54  open consisting design files used for production of first prototypes. 

[image]
[image]
[image]

Luxonis-Brandon, 2021-06-22:
[image]

Luxonis-Brandon, 2021-07-13:
[image]
[image]

Luxonis-Brandon, 2021-07-13:
It's working with DepthAI as well now.
[image]

Luxonis-Brandon, 2021-07-30:
[image]
[image]

Luxonis-Brandon, 2021-08-31:
[image]

Passed 15 hours at 1 meter depth in water with no leaks:
[image]

So this is actually well over IP67 rated (which is 1 meter for 30 minutes without a leak).

Luxonis-Brandon, 2021-08-31:
The most recent PCB fabrication just finished as well:
[image]

Luxonis-Brandon, 2021-09-29:
[image]

Luxonis-Brandon, 2021-10-05:
Latest assembled version:
[image]
Using better screws this time so there's no rust issue when installed outdoors/in-weather:
[image]

dnav1, 2021-10-06:
@Luxonis-Brandon Do you have production files available for these yet? Schematics, mechanical files, updated block diagram, etc? And is this OAK-D-CM4-POE model available for purchase yet?

Luxonis-David, 2021-10-06:
@dnav1 please check the PR #110 for the project and production files, it will be merged when we will get the product fully tested.
The same goes for selling those, we will start selling limited quantity when product is vetted.
For higher quantity orders we need to get more enclosures and PCBAs manufactured first.

adriaan-mernok, 2021-12-06:
@Luxonis-Brandon I don't see a preorder link for this at the top of the issue. Any idea when this will become available for ordering?

adriaan-mernok, 2021-12-06:
@Luxonis-Brandon Do the stereo camera modules on these have IR blockers? Or will one be able to mount an external IR illuminator/laser projector to improve depth perception in low light conditions?

Luxonis-David, 2021-12-06:
@adriaan-mernok those come by default with IR blocking filter on the lens.
For larger customer orders for sure we can assemble the product with NoIR OV9282 stereo cameras, but the custom batch quantity should over 100pcs.
Thoughts?

adriaan-mernok, 2021-12-06:
@Luxonis-Brandon I am currently in the proof-of-concept phase. I assume that the board does support the NoIR OV9282 camera modules by default? Thus I believe that I will just be replacing the module myself at first.

Luxonis-David, 2021-12-06:
@adriaan-mernok yes that is right, all OAK boards does support the NoIR cameras by default, you just need to be careful which one is needed as replacement.
For instance on OAK-D-CM4-POE we do already use Arducam short FPC style 33-pin connector interface.
This is different to OAK-D which uses Sunny style FPC and connectors, you can check the cameras modules are different on above photos, of course sensor is the same OV9282.

rdpeake, 2022-05-24:
Given the changes to the poe pro do you plan to bring the m12 and m8 connectors as well as ir capabilities to this platform in a future revision?

Erol444, 2022-05-24:
Hi @rdpeake ,
Probably not, as CM3/CM4 models will soon be "deprecated" as we will have Keem Bay ([series 3](https://docs.luxonis.com/projects/hardware/en/latest/pages/articles/oak-s3.html)) soon, which has an on-board quadcore ARM running Linux, which means there will be no need for CM3/CM4. And we will have `OAK-D-Pro-PoE-S3` (with Keem Bay) probably by the end of this year. Thoughts?

rdpeake, 2022-05-25:
will the keem bay systems have IO lines for integration with other systems along side the cam and ai control? also will there be any OS support issues between the cm3/4 vs keem bay when it comes to installing other packages - like ROS2 core etc?

GNejc, 2022-05-29:
Hi @rdpeake 

for the IO's on OAK-D-PRO-POE-S3 we are thinking of having it in the same form factor as the OAK-D-PRO-POE (the S2 variant). That means that it will incorporate the same M12 and M8 connector as the OAK-D-PRO-POE that we have now.

closed 2023-03-16 by Erol444