luxonis/oak-hardware #62: Different version of OAK-D ? (intl vs a/b adaptor)

TheBricktop, 2021-05-06:
I've found these two positions of OAK-D, 
https://eu.mouser.com/ProductDetail/Luxonis/OAK-D-AB?qs=DPoM0jnrROWvGlNkjV8g6g%3D%3D
and
https://eu.mouser.com/ProductDetail/Luxonis/OAK-D-INTL?qs=DPoM0jnrROXjQVWBpbQqSw%3D%3D

What is the actual difference between those two ?
Comparision shows different "adaptor" but what does it means? Is it about different adopters or an actual interface to some other part?

Luxonis-Brandon, 2021-05-06:
Hi @TheBricktop ,

Thanks for the interest and sorry about the confusion here.

The only difference is the power supply.  

The **INTL** version has this power supply:

[image]

And the **AB** has this power supply:
[image]

The AB works only in "US Style" plugs - so the US, Canada, Mexico, Japan, Taiwan.  The INTL works in all those countries, and a bunch others.  We've yet to get a complaint about the INTL not working in any of the countries we've shipped to - so as far as we can tell it works in every country.  

We actually have been working for a bit to switch over to only INTL to avoid this sort of confusion - given that it works everywhere - and also acts as a built-in travel adapter for folks in the US (or other AB countries) that travel with their OAK-D.  :-)

Thoughts?

Thanks,
Brandon

TheBricktop, 2021-05-07:
@Luxonis-Brandon thank You for fast response, can OAK-D be powered through the usbC cable? This is due to the fact that we want to use sensor with mobile and egde devices.

closed 2021-06-03 by Luxonis-Brandon