luxonis/oak-hardware #91: Equivalent of OAK-D-IOT-40 with OAK-SOM-PRO

Luxonis-Brandon, 2021-07-16:
### Start with the `why`:

The 1099EMB has a reference design for embedded use cases (actually it has 2, the [OAK-D-IoT-40](https://shop.luxonis.com/collections/all/products/bw1092) and [OAK-D-IoT-75](https://shop.luxonis.com/collections/all/products/oak-d-iot-75), but the 2099 does not yet have such a reference design.  

And in many cases, the 2099 may be more appropriate for such embedded applications - including cases like the [CommuteGuardian](https://discuss.luxonis.com/d/8-it-works-working-prototype-of-commute-guardian), where it may be desirable to store video to onboard eMMC or SD-Card.  

And also in many applications having onboard microphones is quite helpful or an absolute requirement (which the 2099 support).

Another related update (which we should propagate back to the 1092) is that the ESP32 programmer microUSB connector can easily break off (as it's surface mount, and microUSB is just generally too fragile), and also it’s a bit annoying to have to have 2x USB cables plugged into the board.

We also realized that we could put a USB2 hub on board, and allow the USB3 to go straight to the MX, allowing both the ESP32 programmer (microUSB in the current 1092 design) and the Myriad X USB2 interface could be combined into the single USB3C connector.  This will make the development experience easier (and make building an enclosure easier).

### Move to the `how`:

Using the same idea as the 1092, make a 2092 equivalent using the 2099 SOM.
We should also change the CCMs with the new/better design from Arducam, as it allows better hardware-level sync, takes less board space, and is more resilient mechanically (both for production and field robustness).

### Move to the `what`:

- Like the 1092 (small, onboard cameras)
- But with 2099 SOM
- 6 microphones
- Add onboard speaker, mono is fine (maybe second channel to through-hole solder points?)
- Built-in SD-Card connected to 2099 SOM (not ESP32)
- Do the boot-button approach that we’ve done on other embedded designs, with default boot mode being NOR flash (0x03), and when the button is pressed, USB-boot is active instead.
- Combine the ESP32 USB2 and the MX USB2 with an onboard USB hub to have only a single USB connection - the USB3C.
- Use new CCMs from Arducam: 
 - RGB IMX378
 - Stereo OV9282

[image]

Luxonis-Brandon, 2021-07-16:
Status update on this:

Initial design is complete, and we're reviewing to order soon.  
[image]

Luxonis-Brandon, 2021-08-31:
Bare PCBs are fabbed:
[image]

Luxonis-Brandon, 2021-09-16:
[image]

Luxonis-Brandon, 2021-09-17:
[image]

Luxonis-Brandon, 2021-09-25:
Works.
[image]

jamie807, 2021-12-09:
Hi @Luxonis-Brandon , just wondering if there are any plans to make files available for this project? In terms of baseboards for the OAD-SOM-PRO, I could only see files available for NG2094_OAK-D-PRO-W-DEV.

Luxonis-Brandon, 2021-12-09:
Yes.  We may have just forgotten.  CC: @GNejc for adding the files or pointing out where they may be.  I think in a PR if they're up.

GNejc, 2021-12-09:
Hi @jamie807,

We will upload it shortly and sorry for the long wait for this design. Please keep in mind that this is the first prototypes and there were a few major bugs found in the design. That is also the reason why the design has not been uploaded yet.

Before using this design for your projects, please read the README file for changes that need to be implemented.

jamie807, 2021-12-09:
Thanks! I appreciate it!

Luxonis-Brandon, 2022-04-14:
Latest batch is back:
[image]

Luxonis-Brandon, 2022-04-20:
First completed unit is in!

[image]

Luxonis-Brandon, 2022-05-10:
[image]

meyerfy, 2022-12-16:
This looks fantastic! Are you already / planning to sell this model?
Also could you link to the baseboard HW files? Can't seem to find them.
Thanks!!

GNejc, 2022-12-17:
Hi Meyerfy,

sorry to inform you that we will not be selling this model through our shop/distributors. As we do not EOL our products we can still support you in getting the devices however it is associated with MOQ. Let me know if that would be something you are interested in.

You can still find the design files for the base board under NG2092 [here](https://github.com/luxonis/depthai-hardware/tree/master/NG2092_OAK-D-IoT-40)