luxonis/oak-hardware #97: Add OAK-IoT-SoM

Luxonis-David, 2021-07-29:
Add OAK-IoT-SoM specifications instructions etc.

nalzok, 2021-10-29:
Could you please also add the Assembly Drawing and BoM for each SoM variant? I just need a rough idea of the onboard components, but I work on a Mac so I don't have access to Altium Designer.

Luxonis-David, 2021-10-30:
@nalzok unfortunately we can not share the design details about our SoMs, as all hardware design related directly to MyriadX is a proprietary to Luxonis only.

nalzok, 2021-10-30:
@Luxonis-David I see. Could you tell me what this connector is, and what it is for? I think that is an interface instead of a design detail :)

[image]

Luxonis-David, 2021-11-01:
@nalzok yes sure, it is a flat cable connector for JTAG debugging for Myriad X.
This is not exposed to the users and is meant only for internal debugging and development.
Most of the same functionality can be achieved over USB interface though, but we would need to grant you an access in advance,

closed 2023-03-16 by Erol444