luxonis/oak-hardware #99: Android support?

TheBricktop, 2021-08-02:
I know that it might not be on the list now but Is OAK D able to be used with android devices ?

Luxonis-Brandon, 2021-08-02:
Hi @TheBricktop ,

So actually at least one company has now done this (so it's doable).  But they didn't share it back open source, unfortunately.  They actually just compiled DepthAI directly into the Android device, apparently.   

@themarpe can share more details as to our slow march to allowing this directly and open-source-ly.

Thoughts?

Thanks,
Brandon

TheBricktop, 2021-08-02:
I was looking at python for android for compiling but this approach is kinda interesting, so it could be possible to compile the depthAI to AAR or APK and then deploy directly to the device or embed in some graphical engine?
Id be more than happy to see some docker image for that.

themarpe, 2021-08-02:
@TheBricktop with latest work being done on XLink (https://github.com/luxonis/XLink), with regards integrating libusb - we aim to test out a PR that addresses direct support for Android in libusb. I have a positive feeling about the integration, but we'll know more once we go ahead and test it out.

With that working, a APK could be made just like usually, with NDK to compile the DepthAI library.

TheBricktop, 2021-08-03:
@themarpe  i thought that libusb is already supported on Android according to their site https://libusb.info/ ? Does that involve compiling the library or is it more complicated?

themarpe, 2021-08-03:
@TheBricktop Android is supported, but not from "userspace". The underlying backend I think is mostly the same as Linux when compiling libusb on Android system. But access to actual devices (/dev/bus/usb I think) requires higher permissions as Android applications are granted. So its great when an Android system is used with complete root access, there libusb works as is. 
The before mentioned PR (https://github.com/libusb/libusb/pull/874) addresses "userspace" (regular Android applications) usage, by calling Android SDK API to request permissions to a certain USB device and then continue communicating with it as before. 

This was able to be done before as a "hack" by getting the file descriptor from Java by calling the appropriate SDK API and then using that in libusb to retain the rest of libusb API.

With this PR, its made such that this happens behind the scenes without any "hackish" solution of passing fds around, just a call to libusb options, passing in a JavaVM pointer at initialization time.

TheBricktop, 2021-08-04:
So if I would like to compile depthAI for android where should i start? Are there any big hoops ahead?

Luxonis-Brandon, 2021-08-04:
I'm not sure if we have any guides on it.  At least one of our customers has done it - but we have not.

themarpe, 2021-08-05:
@TheBricktop as of right now, without digging deeper, there will most likely be challenges ahead. If you want to take a shot at it, then in a summary, you'll have to do the following:
 - Take a look at the above mentioned libusb PR - use that branch to compile for Android
 - Clone https://github.com/luxonis/XLink
 - Clone https://github.com/luxonis/depthai-core
 - Prepare Android NDK environment with CMake
 - Specify CMake build options for `depthai-core`: `-D DEPTHAI_XLINK_LOCAL=[path/to/cloned/xlink]` `-D LIBUSB_INCLUDE_DIR=[path/to/libusb/include]` `-D LIBUSB_LIBRARY=[path/to/libusb/library]`
 - Modify XLink to be able to pass in a JavaVM pointer to libusb options
 - ...

It involves a lot of separate components as well as integrating everything correctly under Android NDK in the end, so not a straight forward instructions as there will definitively be issues to solve along the way.

If you'd like to "just use it", probably better to wait for us as we'll be doing some improvements in component interaction which will move us closer to be able to integrate DepthAI cleanly in Android NDK as well

TheBricktop, 2021-08-05:
@themarpe 
Well thats most helpfull, mind if I compile the knowledge that You provided and make a repo commited to depthAI on android? (or other edge os like raspbian)

themarpe, 2021-08-05:
Of course, any contribution is greatly appreciated!

There might be some overlap of work but feel free to fork both `XLink` and `depthai-core` and open PRs with any improvements.
Note that XLink will be getting a slight refactor of how libusb is integrated, etc... which could collide a bit with work done before that, but there are still other things to do, which are completely orthogonal.
1. Exposing capability of passing JavaVM pointer to libusb from XLink API (XLink.h -> neat function accepting options for libusb or maybe better yet, a libusb context which can be preinitialized however one sees fit)
2. Exposing capability of passing JavaVM pointer to XLink from `depthai-core` API

Also I think there was some discussion of getting JavaVM pointer directly from libusb in which case the above two points wouldn't be needed, and would make everything much nicer as well. Feel free to check the above mentioned libusb PR for more details on that.

Other OSes like Raspbian are already supported :)

sdhaliwal, 2022-08-18:
Hi, Is there any update on this. We have purchased a number of OAK-D-IOT-75 devices which we would like to connect to android tablets. Is this possible  at this/future time. As @TheBricktop mentioned, the idea is to use Depth ai functionality along with Kivy and python-for-android.

Thanks a lot

themarpe, 2022-08-28:
@sdhaliwal 

Android is supported since version https://github.com/luxonis/depthai-core/releases/tag/v2.16.0

Feel free to check out the following repositories for inspiration of its usage.
 - https://github.com/ibaiGorordo/depthai-android-jni-example
 - https://github.com/ibaiGorordo/depthai-android-unity-example
 - https://github.com/bytedeco/javacpp-presets/tree/depthai/depthai

The last is a template to create Java bindings & build a snapshot for use directly with Android integration. It needs a tad more work however until mainline.

closed 2022-08-28 by themarpe