luxonis/oak-hardware #113: OAK-WD-PRO

Luxonis-Brandon, 2021-08-21:
### Start with the `why`:

We’ve now gotten a bunch of reachouts about something like OAK-D, but with 150 DFOV, IR-laser-dot-projection depth, and matching resolution/FOV from color camera.  

This is similar to this idea, [here](https://github.com/luxonis/depthai-hardware/issues/21), which we already had on the roadmap, but with the change that apparently a lot of applications want this _and_ wide FOV, which is separately [here](https://github.com/luxonis/depthai-hardware/issues/15).

There’ve been so many that it seems like this wide FOV may actually be the majority of the market.  

### Move to the `how`:
Resolving https://github.com/luxonis/depthai-hardware/issues/15, ArduCam now produces an IR-capable OV9282 that seems to match exactly what the market seems to really want, [here](https://www.arducam.com/product/arducam-1mp-ov9282-fisheye-mono-global-shutter-drop-in-replacement-for-depthai-oak-dnoir/).  

We should work with ArduCam to (1) update this OV9282 to the new/standard ArduCam connector format here and (2) make a OV9782 variant with the same connector/format but that blocks IR (for good color representation).

### ​​Move to the `what`:

- 2x OV9282 1280x800 global shutter grayscale 150 DFOV (IR capable)
- 1x OV9782 1280x800 global shutter color 150 DFOV (IR-blocking)
- 2x IR laser dot emitter (for wide FOV)
- 2x IR LED (for wide FOV)
- Similar (same) form-factor as OAK-D-PRO

ptyork, 2021-08-25:
Love this. Given the challenge of resolution/detail at a wide FOV, would it be possible to make just the RGB cam truly modular to allow use of a higher-resolution RGB sensor (e.g., IMX477) instead of the OV9782 for those less concerned with global shutter and matched resolution?

ptyork, 2021-08-28:
> The challenge with modularity is making it still reasonable for calibration to hold/etc.

I guess my thought was that the modularity would be limited only to the RGB camera and not something that was designed to be done frequently. Perhaps simply working with Arducam to standardize the RGB modules' PCB size and mounting it with M2'ish screws instead of adhesive? If so it would seem that a rare/one-time recalibration of the single camera wouldn't be too onerous.

> But perhaps such a design could be useful. Say an aluminum enclosure with M12 lens mounts. Thoughts?

Perhaps. But I _really_ like your bespoke wide angle design concept. Presumably you'd be matching the dual emitters to the 150 degree FOV. So allowing a lens change would seem to be of limited value to the end user. I mean, I can always envision a use, but I don't _think_ that would add value in our case.

A use-case I'm envisioning here is something like off-axis OCR. With a 1MP RGB sensor, would there be enough detail to make out the words on (e.g.) an office placard near the periphery after rectifying a 150 degree fisheye image? I'd imagine the extra resolution would be very useful for this scenario (which for us, might be quite common).

ptyork, 2021-08-28:
Oh, and since we're talking future designs and "pro" models, how feasible would a light-sensor+actuated-IR-filter be? I.e., choose and optimize for active vs. passive depth automatically based on lighting conditions?

Maybe not on the camera modules (bulky/pricey), but integrated into the glass front of an optional "pro+" enclosure, thus making it an upgrade option for either pro model? And if not a near-term option, you could "future-proof" the design for this kind of concept by adding a QT/Qwiic i2c+power connector to the OAK's PCB. Would also allow for other "accessory" enclosures...actuated 2x zoom lens, maybe?? Dunno. Just a thought.

Luxonis-Brandon, 2021-08-30:
Thanks for the brainstorming @ptyork .  So in terms of actuated IR filter - that's an interesting idea.  I will have to think more on the feasibility of the other ideas as well.  Thinking...

Luxonis-Brandon, 2021-09-16:
[image]
[image]
Initial layout of open source design.

ptyork, 2021-09-17:
Very nice. Dumb question: what's that 10-pin connector on the front for? JTAG? Happy to see the BNO085 make the cut on this one.

Luxonis-Brandon, 2021-09-17:
Not a dumb question.  We'll probably change it to be 2 connectors actually.  That is the connector for an external laser-projector board.  We did this so that we can permute on laser dot projector orientations/etc. w/out re-spinning the main board.

Luxonis-Brandon, 2021-10-16:
[image]

doisyg, 2021-10-17:
Looks great! When can we preorder some ?

Luxonis-Brandon, 2021-11-08:
[image]
[image]

These are both initial engineering-sample/experimentation versions.  The final versions will be single-board and smaller.

ptyork, 2021-11-09:
Nice!! Very exciting!

> The final versions will be single-board and smaller.

Fully single-board? IE, no separate SOM? Or as in you're integrating the projectors onto this single board?

Luxonis-Brandon, 2021-11-10:
Thanks @ptyork !  Yes, so what we realized is our customers like having 2 disparate options:
1. Reference designs (like the above) which use the SoM and showcase how to easily/quickly custom integrate.
2. Finished product solutions that are smaller/lighter for cases when customization isn't required.

So then most (perhaps all) product releases will follow this:
1. An open-source reference design which also serves as our first prototypes (as making boards with the SoM is WAY quicker both from engineering standpoint and also fabrication (because they're simple)).
2. A finished product that is chip-down (so can't be open-sourced unfortunately) and single-board which is custom-made for the application.

Thanks again!  -Brandon

axtronical, 2021-12-05:
> Oh, and since we're talking future designs and "pro" models, how feasible would a light-sensor+actuated-IR-filter be? I.e., choose and optimize for active vs. passive depth automatically based on lighting conditions?

I have two cameras for my Raspberry Pi boards that have automatic IR switching based on light level, and it *does* work fast. It has switched to IR as fast as I can turn the light switch in my room off. This is camera uses the 5MP OV5647.

8-Dale

Luxonis-Brandon, 2021-12-11:
Working on the final version of this (non-dev-board version) now that we have validated the concept via the dev board:

[image]

Luxonis-Brandon, 2021-12-24:
We're adding the capability to use USB cables that can fasten to the device.  
[image]

doisyg, 2021-12-25:
That's a great idea !

kota-rr, 2022-02-04:
Hi, 
RGB camera of this module is `OV9782 1280x800 global shutter color 150 DFOV / 127 HFOV (IR-blocking)` which is different from `IMX378 fixed-focus` of https://github.com/luxonis/depthai-hardware/issues/114. 
What are the reasons why hese two models are using different RGB cameras?
Thanks.

Erol444, 2022-02-04:
>What are the reasons why hese two models are using different RGB cameras?
Hello @kota-rr , I believe these will be used only for prototypes, and production devices will actuallyhave the 4k IMX on there with wide lens. Reason is that Arducam currently only has OV9782 color cam with 150 degree FOV.

kota-rr, 2022-02-04:
> > What are the reasons why hese two models are using different RGB cameras?
> 
> Hello @kota-rr , I believe these will be used only for prototypes, and production devices will actuallyhave the 4k IMX on there with wide lens. Reason is that Arducam currently only has OV9782 color cam with 150 degree FOV.

Got it. Thanks.

Luxonis-Brandon, 2022-02-04:
Oh and we’re planning on making it a factory-orders or permutation - whether wide IMX378 or wide OV9782 is wanted.

szabi-luxonis, 2022-02-15:
@doisyg please check: [oak-d-pro-w](https://github.com/luxonis/depthai/tree/oak-d-pro-w) branch. Added support for it. Note that at the moment stereo and RGB resolution must match, and they are hard coded to 720p for the purpose of demo.

richard-xx, 2022-02-25:
@szabi-luxonis 
I run `python3 depthai_demo.py --show left right depth color` using the [oak-d-pro-w](https://github.com/luxonis/depthai/tree/oak-d-pro-w) branch and it's like [DM9098pro](https://docs.luxonis.com/projects/hardware/en/latest/pages/DM9098pro.html#demo), but I don't seem to see similar results. Does it need to be calibrated? Or do I need to be in complete darkness to see the results?

szabi-luxonis, 2022-02-25:
> @szabi-luxonis I run `python3 depthai_demo.py --show left right depth color` using the [oak-d-pro-w](https://github.com/luxonis/depthai/tree/oak-d-pro-w) branch and it's like [...]

Yes, it needs to be calibrated.

szabi-luxonis, 2022-02-25:
> @szabi-luxonis I run `python3 depthai_demo.py --show left right depth color` using the [oak-d-pro-w](https://github.com/luxonis/depthai/tree/oak-d-pro-w) branch and it's like [...]

It needs to be calibrated, but at the moment the demo doesn't support dewarping for left/right frames, which is needed for good depth results. cc: @alex-luxonis @VanDavv

richard-xx, 2022-02-28:
Here is the frame I run the demo:
[image]

This is my device and when I start it up the IR LED doesn't seem to work:
[image]

alex-luxonis, 2022-02-28:
@richard-xx I'll update the demo later today to use the latest IR control API (the low level API doesn't exactly work on OAK-D Pro-W-DEV), and send also mesh files for rectification, as otherwise the depth map won't look good.

Could you try for now this example: 
https://github.com/luxonis/depthai-python/blob/multi_cam_support/examples/StereoDepth/stereo_depth_video.py
Add the option `-lm` to load mesh data.

To enable the IR drivers, add one or both of the lines below:
```patch
 with dai.Device(pipeline) as device:
+    device.setIrLaserDotProjectorBrightness(200)  # current in mA, 0..1200
+    device.setIrFloodLightBrightness(100)         # current in mA, 0..1500
```

slitcch, 2022-03-17:
What's the max framerate of these cameras? 60fps or 120fps?

Erol444, 2022-03-17:
> What's the max framerate of these cameras? 60fps or 120fps?

Hello @slitcch , it depends on which camera module and at which resolution. 

- IMX378 (color cam): 30 FPS at 4k, 60FPS at 1080P
- OV9282 (mono camera): 60 FPS at 720P/800P, 120FPS at 400P

crnolan, 2022-04-22:
Is there a facility to switch the auto-focus camera to manual focus mode? I have an application that requires < 50 cm working distance but the plane of interest is largely fixed in any particular scenario.

Also, does the depth sensing work effectively at distances < 50 cm?

Erol444, 2022-04-22:
Hello @crnolan , 
Yes, you can set the focus to the desired position, [see example here](https://docs.luxonis.com/projects/api/en/latest/samples/ColorCamera/rgb_camera_control/#rgb-camera-control). You can manually change the focus with `,` and `.` keys. 
Thanks, Erik

slitcch, 2022-07-20:
I got my OAK-D Pro W Dev yesterday, and it's pretty good! Here's our hand tracking running with its camera streams: https://twitter.com/mosesturnervevo/status/1549817050859819008 
Some comments:
* Are we sure that that's the maximum brightness that the IR illuminators can strobe at? It's kinda underwhelming compared to Ultraleap's illuminated-IR...
* For my purposes I'd love it if a future generation of the OAK-FFC-4P had FFC connectors for IR dot projectors and floodlights, and FFC modules to match, so that I can change the camera+illuminator placements as I see fit

Luxonis-Brandon, 2022-07-22:
Thanks.  So actually it's possible to add more IR LEDs since the design is open source.  I'm not sure how much illumination Ultraleap has built in.  But since the OAK has AI, CV, depth, and high-resolution/framerate capability built in - it already consumes some USB power.  So adding a bunch of IR LEDs would likely make the power consumption quite high.  

Yes, adding IR interface to the FFC is a great idea.  Will bring that up.  Thanks!

geniusdo, 2022-08-29:
Hi. Does the OAK-D PRO W support fsin I/O? I need the 2 infra cameras to sync with my external imu. I'd appreciate it if you could tell me anything relating fsin I/O support.

moshimeow, 2023-03-13:
We're probably about to buy a few more of these. I'm like 95% sure these do come with IMUs, but is it documented anywhere that they do?

Erol444, 2023-03-13:
Hi @moshimeow ,
Best to check at support@luxonis.com, as it varies across batches - some have BMI270, and some BNO085 IMU.
Thanks, Erik

Erol444, 2023-03-16:
Already released, see [product documentation here](https://docs.luxonis.com/projects/hardware/en/latest/pages/DM9098prow.html)

closed 2023-03-16 by Erol444