luxonis/oak-hardware #114: OAK-D-PRO

Luxonis-Brandon, 2021-08-21:
### Start with the `why`:

#### 1. Mechanical design.  

The mechanical design of OAK-D is limiting and makes it seem intended only for hobby applications, with the following draw-backs: 
 - Only 1/4-20 tripod mount
 - Lack of 2-screw (VESA) for easy panel mounting
 - Bigger than necessary
 - Heavier than necessary

#### 2. Active Illumination

OAK-D was architected for applications where passive depth performs well (and can often outperform passive depth; as IR from the sun can be blocked).  

There are many applications however where

Luxonis-Brandon, 2021-09-22:
It's here!  And everything works!
[image]

Luxonis-Brandon, 2021-09-23:
BELAGO 1.1: [https://ams.com/belago1.1](https://ams.com/belago1.1)

avenarobotics, 2021-09-26:
Good choice, i tested it few days ago. Looks good.

Luxonis-Brandon, 2021-10-11:
> Are you designing OAK-D-PRO hardware to deal with peer sensor interference?

Not explicitly.  That said our multi-camera sync is likely accurate enough that an inherited solution already exists and is likely to be sufficient.  At least for active disparity depth - so for OAK-D-Pro.  (Not sure on ToF.  Will need investigation.)

diablodale, 2021-10-11:
Like https://www.cs.unc.edu/~maimone/media/kinect_VR_2012.pdf

michaelkr, 2021-10-11:
@diablodale  I'm interested in how that might occur. With structured light or Lidar, yes of course, but overlapping illumination patters in stereo systems are expected to *improve* the resulting depth map, not degrade it. Interested, Michael

michaelkr, 2021-10-11:
I don't believe those things should matter - stereo-based systems stereo don't need to associate projected patterns with an individual sensor (or any sensor at all, for that matter). The dots are there just to provide texture in the scene in situations where there is none (uniformly-painted walls, for eg), so that stereo matching can occur - but it does not matter from what source. For example, the Intel RealSense stereo cameras (*not* their structured light cameras) are of a similar design and can overlap without issue.

nalzok, 2021-10-12:
Is there any chance you can center the laser dot projector on the device? Maybe it's just me, but the asymmetric design doesn't seem super elegant. It would be awesome if you could move the projector right below the color camera!

doisyg, 2021-10-12:
> I don't believe those things should matter - stereo-based systems stereo don't need to associate projected patterns with an individual sensor (or any sensor at all, for that matter). The dots are [...]

Same observation here , we have been using d435s with overlapping FoV for a while with no issues. To my understanding, as long as the patterns are sharp enough, they are adding more textures for the stereo matching. The only way I see it could have an adverse effect is if you have enough patterns projected to saturate the camera (or to re-use the spray analogy, if you have painted completely and uniformly your wall).

avenarobotics, 2021-10-15:
What stereo matching algorithm You want to use? What limitation it has? Resolution/disparity range/fps ?

Luxonis-Brandon, 2021-11-10:
Sorry @KySmith1 I missed this.  Likely February 2022 but just a guess right now.

gluxkind-k, 2021-11-23:
> It's here! And everything works!
> [image]

What is the width dimension of this (including the enclosure and without the enclosure)?

Luxonis-Brandon, 2021-11-23:
Great question.  I don't immediately have these - so I'll ask the team for specifics and to share them here.

simondemeule, 2021-12-18:
Hi! Just a curious onlooker chiming in. 

The hardware you are developing is awesome, and there is one specific use case I could see it absolutely shredding: eye-tracking. With the right software, this could possibly match the performance of multi-thousand-dollar research-grade eye-tracking devices. 

Classical algorithms for eye-tracking rely on point light sources creating sharp reflections on the eye; through some geometric formulation of the problem, it is possible to recover eye position very accurately. Depending on the number of cameras and number of light sources, the problem becomes either under or over-constrained, and can be solved with different degrees of accuracy, with or without assumptions about head position. There are also likely some more modern machine-learning based approaches to this problem, but fundamentally, camera and light source count define how the problem is under or over-constrained. Two cameras and one point source is very nice, because it allows accurate inference of head position in 3D space. Most eye-trackers around this price point lack a stereo infrared camera pair; this requires the user to keep their head at a calibrated distance from the eye-tracking device. People need to refrain from moving around naturally for things to work properly, which isn't great for something like interactive art installations.

In all cases, having high resolution and framerate is important for good eye-tracking. It seems like the stereo pair you have fits the bill nicely — the resolution is likely high enough to get a decent patch of pixels on the eyes, and the framerate is great. Another characteristic of higher-end, stereo eye-tracking systems I've noticed is that the spacing between the pairs is often wider than that of the OAK-D. With a bit of maths it would be possible to find an optimal field-of-view and baseline-width for this task, but that would imply making a completely new hardware design, which maybe is a bit much to ask right now.

However, there is one possibly straightforward optional change that could open a lot of possibilities for this application. If it was possible to produce a variant of this that substitutes the RGB-bayer, IR-filtered IMX378 for its infrared-capable, monochromatic variant, this could open up the door to high-resolution pupillometry in dark lighting conditions — the ability to accurately measure pupil diameter in real-time. This can be a very interesting data point for various applications. Since pupil diameter is  influenced by many psychological factors, but also by the amount of light the eye recieves, this is essentially always done in controlled lighting environments that are often dark — hence it makes sense to use infrared.

And also, just a silly question coming from the aesthetics-obsessed part of me: I assume the white rectangle in the prototype between the IR laser and LED is just the PCB — will the production unit be black?

Luxonis-Brandon, 2021-12-20:
Thanks @simondemeule !  Yes, the white part goes away in production.  Here's more what final production will look like:

[image]

[image]
OAK-D-S2 (passive depth and CV) is the top one, and OAK-D-Pro is the bottom one in the image.

Here's OAK-D-Pro by itself:

[image]

And for doing the change you propose.  Yes, we can likely work with ArduCam to do some custom builds of this type.  It should be relatively straightforward to do so.  

Please feel free to reach out to support@luxonis.com (and link to your comment please if you don't mind) to see about the costs involved here to produce the number of devices you would like here.

Thanks again,
Brandon

Luxonis-Brandon, 2021-12-21:
[image]

sukhrajklair, 2021-12-21:
Hi!
Are you planning to build a POE version of this model like the OAK-D-POE?

Luxonis-Brandon, 2021-12-21:
Yes see here:
https://github.com/luxonis/depthai-hardware/issues/142#issuecomment-998975585

[image]

Luxonis-Brandon, 2021-12-24:
We're adding the capability to use USB cables that can fasten to the device.  
[image]

simondemeule, 2021-12-27:
@Luxonis-Brandon awesome! I'm happy to know we might be able to make this a reality. I'll be in touch.

The production units look nice, this is great industrial design!

Luxonis-Brandon, 2021-12-28:
Thanks @simondemeule !  Oh and given that you are into machines that make art, check this out done w/ OAK-D-Lite:

[image]

https://github.com/keijiro/DepthAITestbed

With OAK-D-Pro I'm sure there will be even more interesting things that can be done.  :-).

melsanharawi, 2022-01-06:
Hi guys,

Seems to be a wonderful project. Since depth perception should be more accurate with this new version, do you think it could be used for 3d scanning ? It was anounced 3% depth error for the oak-d, do you have an idea of the new depth accuracy in this version ?

Thanks a lot

Erol444, 2022-01-08:
Hello @melsanharawi , so the below 3% depth error rate is when you have near perfect condition, so great lighting and good texture. When you don't have these, it can be a lot worse (completely random), since disparity matching can't do it's job. Pro having laser dot projector helps with that, and so the error rate is still below 3%.
Thanks, Erik

Luxonis-Brandon, 2022-01-08:
We also now have ToF for higher accuracy depth for scanning applications.

simondemeule, 2022-01-08:
@Luxonis-Brandon whoa! That is huge! Is this on a different hardware platform using dedicated ToF image sensors, or are you somehow able to extract accurate-enough timing information from this hardware to create ToF depth maps?

Unrelated to this, I've been a bit slow getting in touch with ArduCam but will do shortly — I would really love to see this project idea come to fruition.

michaelkr, 2022-01-09:
> We also now have ToF for higher accuracy depth for scanning applications.

Any additional details on this? What sensor is providing this, or the resolution...seems like a large feature add, I'm fascinated!

themarpe, 2022-01-11:
Hi @simondemeule and @michaelkr 
Check the following: https://docs.luxonis.com/projects/hardware/en/latest/#modular-cameras-designs and specifically the ToF: https://docs.luxonis.com/projects/hardware/en/latest/pages/DM0255.html

To answer your question Simon, this is separate ToF sensor, which can be used with our "FFC" lineup. We don't yet have an off the shelf product integrating it, but that might change in the future.

BZandi, 2022-03-20:
> Hi! Just a curious onlooker chiming in. 
> 
> 
> 
> The hardware you are developing is awesome, and there is one specific use case I could see it absolutely shredding: eye-tracking. With the right software, this could possibly match the performance of multi-thousand-dollar research-grade eye-tracking devices. 
> 
> 
> 
> Classical algorithms for eye-tracking rely on point light sources creating sharp reflections on the eye; through some geometric formulation of the problem, it is possible to recover eye position very accurately. Depending on the number of cameras and number of light sources, the problem becomes either under or over-constrained, and can be solved with different degrees of accuracy, with or without assumptions about head position. There are also likely some more modern machine-learning based approaches to this problem, but fundamentally, camera and light source count define how the problem is under or over-constrained. Two cameras and one point source is very nice, because it allows accurate inference of head position in 3D space. Most eye-trackers around this price point lack a stereo infrared camera pair; this requires the user to keep their head at a calibrated distance from the eye-tracking device. People need to refrain from moving around naturally for things to work properly, which isn't great for something like interactive art installations.
> 
> 
> 
> In all cases, having high resolution and framerate is important for good eye-tracking. It seems like the stereo pair you have fits the bill nicely — the resolution is likely high enough to get a decent patch of pixels on the eyes, and the framerate is great. Another characteristic of higher-end, stereo eye-tracking systems I've noticed is that the spacing between the pairs is often wider than that of the OAK-D. With a bit of maths it would be possible to find an optimal field-of-view and baseline-width for this task, but that would imply making a completely new hardware design, which maybe is a bit much to ask right now.
> 
> 
> 
> However, there is one possibly straightforward optional change that could open a lot of possibilities for this application. If it was possible to produce a variant of this that substitutes the RGB-bayer, IR-filtered IMX378 for its infrared-capable, monochromatic variant, this could open up the door to high-resolution pupillometry in dark lighting conditions — the ability to accurately measure pupil diameter in real-time. This can be a very interesting data point for various applications. Since pupil diameter is  influenced by many psychological factors, but also by the amount of light the eye recieves, this is essentially always done in controlled lighting environments that are often dark — hence it makes sense to use infrared. The resolution and field of view of the stereo cameras would make it challenging to derive good pupil diameter estimates, however the central camera matches up what is seen in some research-grade setups, given it is placed close enough to the user.
> 

@simondemeule If you like to do high resolution pupillometry with a limited budget, the PupilEXT GitHub repo could be also of interest for you: [https://github.com/openPupil/Open-PupilEXT](https://github.com/openPupil/Open-PupilEXT)

Erol444, 2023-03-16:
Already released, see [product documentation here](https://docs.luxonis.com/projects/hardware/en/latest/pages/DM9098pro.html)

closed 2023-03-16 by Erol444