luxonis/oak-hardware #115: OAK-D-LR

Luxonis-Brandon, 2021-08-21:
### Start with the `why`:

Improve the mechanical of OAK-D, as being done in https://github.com/luxonis/depthai-hardware/issues/114 but without adding the cost of active depth (both in terms of $, and in terms of worse bright outdoor depth performance).

### Move to the `how`:

Reuse all of OAK-D-PRO but no-populate the active illumination components, black out the glass where these would be, and use the existing OAK-D CCM.  

### Move to the `what`:

Make an existing passive-only OAK-D successor with the mechanical advantages of OAK-D-PRO.

Erol444, 2021-11-21:
Some more images
[image]
[image]

Luxonis-Brandon, 2021-12-16:
OAK-D-S2 (top)
OAK-D-Pro (bottom)
[image]

Luxonis-Brandon, 2021-12-24:
We're adding the capability to use USB cables that can fasten to the device.  
[image]

jcrupi, 2022-03-25:
Has IMU?

Luxonis-David, 2022-03-25:
@jcrupi yes it does.

Luxonis-Brandon, 2022-05-07:
[image]
https://shop.luxonis.com/collections/usb/products/oak-d-s2

Shipping to our warehouses now.

closed 2022-05-07 by Luxonis-Brandon