luxonis/oak-hardware #142: OAK-D-PRO-POE

GNejc, 2021-10-02:
Start with the `why`:
OAK-D-POE was our “make it exist” product.  It served as a pathfinding design to see what is the community's reaction to it and what could be improved. The OAK-D-PRO-POE will be the next step and an upgrade to the original OAK-D-POE, implementing user feedback and improvements through experience gained from previous design.

What we’ve seen so far on all models is that customers do care quite a bit about:
- Weight
- Size
- Cost
- IP67 sealed
- Active Illumination
- IR laser dot projection
- blanket IR LED illumination

The OG OAK-D-POE hit 4 (IP67) fine, but is heavy, large, and fairly expensive to make.

Move to the `how`:

Connectivity and power:
Just like the OAK-D-POE design, the OAK-D-PRO-POE will have and ethernet connection for data and PoE for power. But that is where similarities end, as the OAK-D-PRO-POE will feature a more robust and industrial connector type called M12.

Form factor:
Form factor will be heavily based on the OAK-D-PRO design with it's sleek small case. In the quest for "as small as possible"  making the design chip down (MX chip on the main board and not on the SOM) makes a lot of sense, as we loose two large connectors that are used on the SOMs. In addition to chip down design, optimization of the POE power delivery circuit will also save a lot of space, so we are opting to go from a discrete solution (OAK-D-POE) to a module design. And the design will have the same IP67 rating as was with OAK-D-POE

Other features:
Implementing IR led and dot projector will be the same as in the OAK-D-PRO model, as it is already in the testing phase and shows a lot of promise.

Move to the `what`:

A small, lightweight, cost-reduced version of OAK-D-POE that is still IP67 sealed, that has IR laser dot projector and IR illumination LED on-board (Pro version).

Luxonis-Brandon, 2021-11-09:
CAD Comparison to OAK-D PoE below:
[image]
[image]
[image]

Luxonis-Brandon, 2021-12-13:
We're going to do a wide FOV variant of OAK-D Pro-PoE as well.  Same formfactor, just slightly different front cover.

[image]

artificiel, 2021-12-14:
what is the expected timeframe for the OAK-D Pro-W-PoE? 

thanks!

Luxonis-Brandon, 2021-12-14:
Hi @artificiel ,

We'll likely have these OAK-D Pro-W-PoE in March.  We would have them in January but likely these will be pushed by an extended (because of COVID) shutdowns in January/February in Asia where these are produced.  

So March is a good guesstimate.  We can add these to the Beta Store if you'd like to be able to pre-order to get some of the first prototype finished units.  Just let us know.  :-)

https://shop.luxonis.com/collections/beta-store

Thanks for the interest.

Luxonis-Brandon, 2021-12-16:
Comparison of first prototypes of OAK-D Pro-PoE with OAK-D PoE
[image]

Luxonis-Brandon, 2021-12-21:
[image]

Luxonis-Brandon, 2021-12-21:
[image]
[image]

ShawnSchaerer, 2022-01-06:
What will the temperature range be on this camera?

GNejc, 2022-01-06:
Hi ShawnSchaerer,

we have not yet done the thermal/stress testing on the design (we just recently got the boards), but we are expecting the unit to be able to have a temperature range from 0°C - 70°C at least.

ShawnSchaerer, 2022-01-06:
It would be great If you could hit 80 or 85C
> Hi ShawnSchaerer,
> 
> we have not yet done the thermal/stress testing on the design (we just recently got the boards), but we are expecting the unit to be able to have a temperature range from 0°C - 70°C at least.

GNejc, 2022-01-06:
Those are more conservative numbers, but we are shooting for 85C.

ShawnSchaerer, 2022-01-06:
Great.  Industrial temperatures are what we are looking for

robotwhispering, 2022-01-18:
What wavelength laser are you using? Are you keeping it near 832nm to mitigate blooming from the sun? 
Are you planning any options for something more powerful than a class 1? Maybe 3R?

Erol444, 2022-01-18:
Hello @robotwhispering , it's near 940nm (projector specs[ here](https://docs.luxonis.com/projects/hardware/en/latest/pages/DM9098.html#projector-specifications)). We don't have any plans for stronger lasers, as even with this one dots are quite visible at +15m. What would be the application for more powerful ones?

Luxonis-Brandon, 2022-01-24:
[image]

sam598, 2022-02-23:
> * M8 connector for GPIO, hardware multi-device sync, and direct external power (instead of PoE).
> 
> ##### We intended to put these on the M8:
> * power input (5V) as an option instead of PoE, or 5V power output for some external circuitry
> * USB2 (D+ and D-)
> * camera IOs: FSIN (frame sync) and STROBE (for driving a flash)
> * 2 other aux IOs, capable of UART, I2C or GPIO

Is there additional information on the FSIN (frame sync) feature?

- What is the voltage/duration/type of signal for triggering the cameras?
- Will the frame sync be able to trigger both the rolling shutter RGB and global shutter IR cameras?
- What will be the tolerance of the frame sync signal? Does it have to be at a set hertz/framerate? Or can it be triggered arbitrarily as long as there is enough time between frames?

GNejc, 2022-02-25:
Hello sam598,

sorry for a late response. 

> What is the voltage/duration/type of signal for triggering the cameras?

- As we have a buffer on this signal, the maximum voltage tolerance of the pin is 5V.
- There is a minimum on time for FSIN signal and depends on the clock provided to the camera. The range of the min on time is between 78ns - 833ns (@alex-luxonis can correct me if wrong here)
- You need to apply a rising edge to the FSIN pin to start the image capture. 

> Will the frame sync be able to trigger both the rolling shutter RGB and global shutter IR cameras?

The FSYNC signal is routed to both Stereo cameras, RGB camera and also routed to the MyriadX VPU.

> What will be the tolerance of the frame sync signal? Does it have to be at a set hertz/framerate? Or can it be triggered arbitrarily as long as there is enough time between frames?

@alex-luxonis would be able to answer this? 

**NOTE**!!:
I just want to point out that we will be actually changing the M8 pinout in the next boards revision, to have a more robust and standard interface, that the industry is using.

New pinout will be:
- power input (5V) as an option instead of PoE, or 5V power output for some external circuitry
- USB2 (D+ and D-)
- Opto isolated FSYNC input ([input circuitry reference](https://docs.baslerweb.com/circuit-diagrams-(ace)#opto-coupled-input-line))
- Opto isolated Strobe output ([output circuitry reference](https://docs.baslerweb.com/circuit-diagrams-(ace)#opto-coupled-input-line))
- GPIO

With regards to the above questions the major change will be the Input voltage tolerance which would now be 5V - 24V.

Let me know if you have any further questions.

Nejc

YoshiyasuIzumi, 2022-03-25:
Hello @GNejc and @sam598,
I'm really interested in this feature.
> What will be the tolerance of the frame sync signal? Does it have to be at a set hertz/framerate? Or can it be triggered arbitrarily as long as there is enough time between frames?

Related to this topic, I need more information on the following two points.
- Is arbitrary timing trigger supported? 
- Does OAK-D Pro-PoE provide trigger mode for burst multiple frame predefined frame rate with color and depth sync?

Any additional information is helpful, thanks!

alex-luxonis, 2022-03-28:
In general two operation modes are possible where an external sync/trigger signal is used:
1. Continuous streaming with external frame sync input (also named in sensor documentation: "slave" mode) - the sync signal has strict timing requirements (usually in the range of few microseconds), its rate needs to follow closely the configured sensor frame rate, otherwise would cause the sensor to output a bad frame. We have a diagram of the sensor operations here: https://docs.luxonis.com/projects/hardware/en/latest/pages/guides/sync_frames.html?h#synchronizing-frames-externally
For the global shutter OV9282(mono) and OV9782(color), it's possible to configure the period of the external signal also as a multiple of the sensor frame period, so for example: sensor 30Hz, external fsync 10Hz. But not clear if the same applies to IMX378, we'll have to test.

2. Snapshot trigger mode. Supported at least by OV9282/OV9782 (we need to confirm for IMX378, or try to emulate it from firmware), the sensor stays in idle, then when it receives a pulse on the pin, it starts streaming a burst of frames (configurable), at the preconfigured rate (`.setFps(..)`). Note that for OV9282/OV9782, the first frame in the group has degraded quality, and depending on application's needs, it may have to be discarded.

YoshiyasuIzumi, 2022-04-11:
@alex-luxonis Thank you for sharing great insights. I can't wait to try OAK-D Pro PoE! If you have update for IMX378 also support Snapshot mode, please share us!
I have one more question. Would you mind to share us the recommended I/O for OAK-D Pro PoE? I assume M8 female cable(8 pin) is the one but I want to start with reliable option. So which model or product would be recommended?
Thank you!

GNejc, 2022-04-13:
@YoshiyasuIzumi 

sorry for the late reply,

In the latest board revision we are switching the M8 connector from Male to Female and thus would require a new cable as well.
In addition we are also changing the pinout of the M8 connector. The new pinout will have:
- STROBE output and FSYNC input be optically isolated (same type of IO as [Basler](https://docs.baslerweb.com/circuit-diagrams-(ace)#opto-coupled-input-line))
- 1x GPIO with the capability of booting to USB with a pull up resistor. This IO is the same as it was with the previous revision.

As this is currently in the development/testing stages, we do not have any functional devices to attach on these yet. We have developed a testing board which has the signal pins exposed on a header for testing porpoises. We could arrange for a board to be sent along with your OAK-D-PRO-POE.

[image]

[image]

Let me know if you have any other questions.

YoshiyasuIzumi, 2022-04-15:
@GNejc 
Thank you for your reply. Please let me confirm my understanding. 
I thought I only need to connect I/O cable from OAK-D-PRO-POE to signal source(5V) in order to provide a trigger to the camera. To provide the trigger to the camera, do I need to use this "testing board" for this purpose?
And one more thing. I checked [Basler's M8 I/O cable](https://www.baslerweb.com/en/products/cable/basler-power-i-o-cable-m8-6p-open-p-5-m/) and it seems to have only 6 pins. Does this cable match to the pins on the camera? 
Thanks!

robotwhispering, 2022-04-15:
I can help answer that question. The older, Ace line of Basket Basler cameras used a 6 pin cable with Hirose connectors. They had maybe 4 different configurations of cable, depending on if you were connecting with PLC, or if you were powering the camera from the IO port, or POE, or using GPIO vs Opto-isolated IO.

When they moved forward with the Ace2 series, they changed to the 8 pin M8, and use the same cable regardless of how you're intending to connect. If I recall correctly, they also got rid of the GPIO in favor of more Opto-isolated IO. No big deal, but I need to use a pull-up resistor (or pulldown) so that the signal didn't just hover in a near constant state of charge. You either have to pull it up to 1V (1.5?) when you want it to go high, or sink it to 0 when you want a low signal. I can't remember what ohm resistor I used for my control signal, but they have documents for each camera that give you the pinout color code, as well as what resistors would keep the current low enough at each voltage step.
But to be honest, the documentation from Lucid, for their Triton line of Gige cameras has even better documentation, and it's basically the same camera as the basler ace2.
I was controlling heater resistors and wipers for the enclosures that the cameras were in.

In other news, I think I still have some of those 6 pin cables with the Hirose 6 pin connector. If you could put it to good use, I could send you one to test with. Just let me know what your setup is.... What you're connecting and controlling so I give you the most appropriate cable.

YoshiyasuIzumi, 2022-04-21:
Hi [robotwhispering](https://github.com/robotwhispering),
Sorry for my late response and thank you for your insighs!
In my configuration, original trigger logic level is 24V. So I need to convert it to 5V logic level. 
To do that, I'm thinking of using SSR or Isolated photocoupler. For this scenario, if I choose 6 pins M8 cable, I should pull-down(or pull-up) the trigger signal for avoiding channel float. Is my understanding correct?
But I found M8 cable with 8 pins in Lucid vision labs from your reference document, so I guess I shall start with this option.

Please let me know if you have any other comment on this.
Thank you.

Erol444, 2023-03-16:
Already released, see [product documentation here](https://docs.luxonis.com/projects/hardware/en/latest/pages/NG9097pro.html)