luxonis/oak-hardware #145: Enclosure for OAK-D-IoT-75

Luxonis-Brandon, 2021-10-15:
### Start with the `why`:

From the O.G. OAK-D KickStarter we learned that aint nobody want an OAK-D w/out an enclosure.  "To add insult to poorly planned injury" was the most memorable quote about OAK-D-IoT-75 not including an enclosure.  

So, we're making an enclosure for it.

### Move to the `how`:

It's got WiFi.  That's why we didn't make an enclosure at first.  As it was a time-crunched KickStarter delivery.  And we didn't want to be late.  

So we took the time to iterate/test and design in plastic openings that are sufficiently wide for the WiFi to escape.

### Move to the `what`:

And enclosure that has all the things we learned folks want:

- 2x M4 for VESA-mount compatibility.
- 1x Tripod (1/4-20) for desk convenience
- USB3 for DepthAI to host.
- microUSB for direct ESP32 programming.

It's pretty cool looking too:
[image]
[image]
[image]
[image]

Note that this is an older prototype, so the mounting on the back is now M4 and 7.5cm spacing, instead of whatever is shown there (likely M2 at like 43.5cm).

Luxonis-Brandon, 2021-10-15:
This is in tooling now.  Likely will be ready in December.

gizmo84, 2023-05-21:
I'm a little bit late to the party, but it is possible to have decent shipment to Europe given that I bought the original oak-d-wifi on kickstarter ? :)

Erol444, 2023-05-21:
Hi @gizmo84 , I am not sure about the decent shipment part, but we do have it on our shop: https://shop.luxonis.com/products/oak-d-iot-75-enclosure
I hope it helps!