luxonis/oak-hardware #162: How to open up the 3D model files of these cameras?

zmxu, 2021-11-18:
Hi,

We are working on the mechanical design which has OAK-D as a component. However, we could not open the 3D model file or the mechanical file. We have tried several different software, even online tools, but still cannot open them.  What is the exact software that we should use? Thanks.

Luxonis-Brandon, 2021-11-18:
Odd.  Not sure as I always struggle with STEP files.  Asking team.

Erol444, 2021-11-18:
Hello @zmxu , I was able to open the [OAK-D enclosure file](https://github.com/luxonis/depthai-hardware/tree/master/BW1098OAK_USB3C/Mechanical) with [FreeCad](https://www.freecadweb.org/):
[image]

sungtzuwen, 2021-11-19:
Hi @zmxu, not sure which software you used?  I believe the CAD file could be open by Pro/E and Solidworks.

eric-schleicher, 2021-12-07:
@zmxu if you ever need to open/view Step file and save to other (mostly mesh) formats, take a look at CadAssistant.

https://www.opencascade.com/products/cad-assistant/

free to use from the folks who write the geometry kernel under freeCAD.