luxonis/oak-hardware #163: OAK-D Lite Not Working

seahcj, 2021-11-22:
DepthAI DEMO is not working on OAK-D Lite, and here is the log:

```
(.venv) PS D:\_depthai\depthai> python .\depthai_demo.py
Using depthai module from:  D:\_depthai\.venv\lib\site-packages\depthai.cp39-win_amd64.pyd
Depthai version installed:  2.11.1.0.dev+dfd52ac01c3b7514e85cb894b9d5381e999859df
Depthai development version found, skipping check.
Setting up demo...
Available devices:
[0] 1844301061514DF500 [X_LINK_UNBOOTED]
USB Connection speed: UsbSpeed.HIGH
Disabling depth...
Disabling depth preview...
Disabling depthRaw preview...
Disabling left preview...
Disabling rectifiedLeft preview...
Disabling right preview...
Disabling rectifiedRight preview...
Enabling low-bandwidth mode due to low USB speed... (speed: UsbSpeed.HIGH)
Creating MJPEG link for ColorCamera node and color xlink stream...
[1844301061514DF500] [8.400] [system] [error] Attempted to start Color camera - NOT detected!
Stopping demo...
=== TOTAL FPS ===
  File "D:\_depthai\depthai\depthai_demo.py", line 536, in run
    self.instance.run_all(self.conf)
  File "D:\_depthai\depthai\depthai_demo.py", line 59, in run_all
    self.run()
  File "D:\_depthai\depthai\depthai_demo.py", line 245, in run
    self.loop()
  File "D:\_depthai\depthai\depthai_demo.py", line 272, in loop
    self._pv.prepareFrames(callback=self.onNewFrame)
  File "d:\_depthai\depthai\depthai_sdk\src\depthai_sdk\managers\preview_manager.py", line 113, in prepareFrames
    packet = queue.tryGet()
Communication exception - possible device error/misconfiguration. Original message 'Couldn't read data from stream: 'nnInput' (X_LINK_ERROR)'
file:///D:/_depthai/depthai/gui/views/CameraPreview.qml:57:13: TypeError: Cannot read property 'height' of null
file:///D:/_depthai/depthai/gui/views/CameraPreview.qml:56:13: TypeError: Cannot read property 'width' of null
(.venv) PS D:\_depthai\depthai>
```

The error is at the line

```
[1844301061514DF500] [8.400] [system] [error] Attempted to start Color camera - NOT detected!

```

alex-luxonis, 2021-11-22:
@seahcj Sorry for the trouble here, the camera was not detected as the previous library didn't have support for OAK-D-Lite.
I just merged it: https://github.com/luxonis/depthai/pull/525 - could you please pull the latest.

We'll create today a `depthai` release as well, for all other examples (from depthai-python, depthai-core) to work by default on the `main` branch.

Also I see in your logs above `USB Connection speed: UsbSpeed.HIGH`. Looks like the connection is established as USB2, and not USB3 (`SUPER` speed), so the performance may be lowered. Could you check the USB cable and host USB ports.

seahcj, 2021-11-22:
Yes. the main branch is working for OAK-D Lite now. Thanks!

closed 2021-11-22 by seahcj