luxonis/oak-hardware #164: OAK-D Lite Dead Pixel Row

seahcj, 2021-11-22:
My OAK-D LIte has a 'dead pixel row' on the right camera. Is this expected? 

[image]

Thanks

Luxonis-Brandon, 2021-11-22:
Hi @seahcj ,

Sorry about the trouble here.  We had seen this in some failed initial prototype samples but not after that.  Checking with the team now.  

Just to be sure - are there any conditions in which this goes away?  My understanding from what you are seeing (and what we saw from some initial failed prototypes) is that it is there permanently.  

If that's the case, we will likely need to wait for our larger batch to finish to replace your unit.  

Sorry about the trouble here.

And thanks for the backing!

-Brandon

seahcj, 2021-11-23:
I just double-check on the conditions at which the horizontal line appears.

- Cooling down the board doesn't help
- The dead pixel line appears when the background is 'darker'

[image]

Could you make announcement here or somewhere when new larger batch is ready for replacement? thanks.

Luxonis-Brandon, 2021-11-23:
Very interesting.  Thank you for the data.  We're discussing offline now.  Yes, we will make an announcement on KickStarter  when the next batch is ready.

But we can also just ping you here and/or if you shoot us an email at support@luxonis.com we can just submit an order now for a replacement - such that the new unit will ship to you when they are ready.

Thanks and sorry about the trouble here.  

-Brandon

alex-luxonis, 2021-11-23:
@seahcj Could you check how the position of this artifact changes with varying the camera exposure.

The above resolution is cropped (by default) to 640x400. Optionally can run at full 640x480 resolution with:
`python3 depthai_demo.py -monor 800`
(we'll add a proper 480 option as well, but the above should default to 480 on camera anyway)

Then can apply manual ISO and exposure as below. Both ISO and exposure need to be set.
ISO range: 100 - 1600.
Exposure range: 20 - 33300 (us), but would be useful to test with the range 23000 - 33300.

After changing the fields, press `Enter` for the change to take effect. Can also navigate between fields with `Tab` / `Shift` `Tab`.

[image]

Luxonis-Brandon, 2021-11-23:
(Thanks, Alex!  I was going to mention that I wasn't 100% sure that it was a hardware issue yet - but perhaps just some configuration thing.)

seahcj, 2021-11-23:
The horizontal line disappeared in the following setting

[image]

If we don't set the resolution, the line is there

[image]
[image]

By using default setting, and piece of white paper

```
(.venv) PS D:\_depthai\depthai> python .\depthai_demo.py
Using depthai module from:  D:\_depthai\.venv\lib\site-packages\depthai.cp39-win_amd64.pyd
Depthai version installed:  2.13.0.0
Setting up demo...
Available devices:
[0] 1844301061514DF500 [X_LINK_UNBOOTED]
USB Connection speed: UsbSpeed.SUPER
```

Sometime white line is showing

[image]

Sometime is not

[image]

alex-luxonis, 2021-11-23:
Thanks for the tests.

Yes, with auto-exposure we also have anti-banding enabled (trying to prevent flickering due to mains switching, for some type of lights), and it's by default set for 50Hz (Europe) - mains period of 20ms, so auto-exposure sets the exposure in multiple of 10ms. For 30ms the artifact appears at that position, for 20ms it's not shown. So it may depend on the amount of lighting.