luxonis/oak-hardware #170: Schematics & PCB

adriaan-mernok, 2021-12-05:
Hi, are the schematics and PCB for this project also open-source as many of the other projects. I can only find the NG2094_OAK-D-PRO-W-DEV repository, but sadly it is not a single-board assembly as the OAK-D-PRO product is.

I assume this is to protect the IP in the SoMs?

Luxonis-Brandon, 2021-12-06:
To which product are you referring @adriaan-mernok ?  Are you looking for OAK-D-Pro?

If so, you can find all the building blocks from these 2 projects:

1. https://github.com/luxonis/depthai-hardware/tree/master/BW1098OAK_USB3C#bw1098oak-usb3c-baseboard
2. https://github.com/luxonis/depthai-hardware/tree/master/NG2094_OAK-D-PRO-W-DEV#oak-d-pro-w-dev-baseboard-wip

And those are all the pieces necessary to make an OAK-D-Pro.  We normally make an open-source design for every product we make, but given that OAK-D-Pro's only difference from OAK-D is the addition of the IR LED and IR Laser, which are fully open-sourced in `OAK-D-PRO-W-DEV` we figured doing so was a bit redundant.

And in terms of not being able to release chip-down Myriad X designs - this is one of the reasons that we made the DepthAI SoMs.  

- OAK-SoM: https://docs.luxonis.com/projects/hardware/en/latest/pages/BW1099.html#oak-som
- OAK-SoM-IoT: https://docs.luxonis.com/projects/hardware/en/latest/pages/BW1099EMB.html#oak-som-iot
- OAK-SoM-Pro: https://docs.luxonis.com/projects/hardware/en/latest/pages/BW2099.html#oak-som-pro

We wanted as much as possible of our ecosystem to be as open source as possible, as what drives us is `engineering efficiency` and having open-source allows clever engineers to use what we've made in ways we can't anticipate - autonomously and efficiency.  So in the spirt of the (many) answers to the question "Why do cities get more efficient as they get bigger while companies get less efficient" we open-source as much as possible - as it allows such city-efficiency-of-scales effects vs. the inefficiencies associated with big companies.  

And there are multiple constraints on how far we can open source things.  And the first one is the one that, on its face, made it seem impossible to open source anything around the Myriad X:

a. The Myriad X itself is required by Intel to remained closed source.  Even the footprint itself is considered proprietary - and is not allowed to be made public.  

So this constraint alone make it not possible for us to open source any design with the Myriad X onboard.  And it is one of the reasons we came up with the SoM approach.  But there are more reasons:

b.  The Myriad X is a fully-packed, VFBGA ~400-pin package with 0.4mm spacing (center-to-center) between the balls that requires HDI integration present in iPhones/etc.  Such design principals are uncommon for even seasoned Electrical Engineers and require engineers who have made iPhones (or comparable Android devices) or have worked on comparable HDI (laser-via, burried/blind vias, etc.).  And this itself is a huge barrier to entry on such a part.  So this is another reason we made the SoM series.

c.  To facilitate a way to have a HUGE portion of what Luxonis produces in software/firmware/AI-training/deployment/etc. be open source, we need a way to monetize.  Otherwise this is an evanescent experiment that provides no long-term value to the world.  And what we settled on, to enable this, is that we're "a software company that monetizes on hardware."  So even if `a` and `b` above are ignored, this is another independent reason why we sell the SoM (and other closed-source products) like the OAK-D-Pro - it's the only way we monetize to pay back all the open source code we generate and release permissively (as MIT-Licensed) to the world.  

d. There's a huge trade-space here.  The permutations of image sensor type, number of image sensor, lens type, baseline between image sensors, view angles, spectral filtering, etc. all make it such that there isn't a one-size-fits all.  So it's extremely helpful to be able to prototype very quickly.  And even if a company has the capability to produce designs as in `b`, the nature of HDI production is that the machines that produce the boards are SUPER expensive - so any manufacturer of such boards must keep them constantly producing to pay off their investment and justify the continued capital expense.  Which is to say, HDI boards often take 2-5x longer to produce than non-HDI boards when doing standard-turn orders.  And so this is a reason for the SoM approach as well: As it allows the boards that permute across these degrees of freedom to be standard, non-HDI boards which can then be turned by standard board houses - and in rush cases - this allows same-day or next-day board fabrication and assembly by board houses that specialize in fast-turn, low-complexity boards.  So this SoM approach allows such degrees of freedom.  And in fact we even use it for all our new product development because we can iterate so fast.  A good example of this is the OAK-D-Lite first couple versions, which were all SoM-based: https://github.com/luxonis/depthai-hardware/tree/master/DM1095_OAK-D-LITE-DEV_DepthAI_USB3C#oak-d-lite-dev-usb3c-baseboard

So that's the exhaustive list.  

With all that said, please feel free to reach out to me at brandon at luxonis dot com if you are needing chip-down.  We can do so on a per-customer basis as long as this customer is approved by Intel (i.e. it's not NVIDIA trying to copy).

Thanks,
Brandon

Luxonis-David, 2022-05-13:
@adriaan-mernok can we close this issue, have you got the answers you were looking for?
Thanks.

adriaan-mernok, 2022-05-13:
Hi @Luxonis-David. My apologies, it completely slipped my mind to close this. @Luxonis-Brandon, thanks for the insightful response!

closed 2022-05-13 by adriaan-mernok

Luxonis-David, 2022-05-13:
@adriaan-mernok no worries just checking so we don't close the issue if not being sorted out.
Thanks for the quick action.