luxonis/oak-hardware #182: please clarify OAK-D-Lite USB feature-level

diablodale, 2022-01-12:
Hi, at https://github.com/luxonis/depthai-hardware/blob/master/docs/source/pages/DM9095.rst I request text be added to declare the USB feature level for OAK-D-lite. Is it 3.1, 3.2, gen1, gen2, etc.

As comparison, OAK-D docs at https://github.com/luxonis/depthai-hardware/blob/master/docs/source/pages/BW1098OAK.rst clearly states twice "USB 3.1 Gen1 (Type-C)".

This knowledge is requested so that coding+testing of app correctly bounds the USB feature-level based on the specific Luxonis device.

Luxonis-Brandon, 2022-01-12:
Thanks great idea/request.  CC: @Luxonis-David to provide this exactly.

Luxonis-David, 2022-01-13:
PR #183

Luxonis-David, 2022-01-28:
@diablodale do you think we should close this issue based on the PR above?

diablodale, 2022-01-28:
Yes, that's super. Looks good to me on live site. ;-)

closed 2022-01-28 by diablodale