luxonis/oak-hardware #188: Which OAK-D-Lite (Part No. A00483)?

gmacario, 2022-01-16:
Hello,

A few days ago our team received three OAK-D-Lite(s) (Part No. A00483) to be used for Phase 2 of the [OpenCV Spatial AI Contest](https://opencv.org/opencv-spatial-ai-contest/)

https://twitter.com/gpmacario/status/1480612900502917122

I assume the hardware documentation for this specific Part No. is the one available under https://github.com/luxonis/depthai-hardware/tree/master/DM1095_OAK-D-LITE-DEV_DepthAI_USB3C, correct?

If so, what product is the one under https://github.com/luxonis/depthai-hardware/tree/master/DM9095_OAK-D-LITE_DepthAI_USB3C?

Erol444, 2022-01-16:
Hello @gmacario , 
documentation for OAK-D-Lite can be found[ here](https://docs.luxonis.com/projects/hardware/en/latest/pages/DM9095.html). The first link you shared (OAK-D-Lite-DEV) is our prototype version that uses [OAK-SOM](https://docs.luxonis.com/projects/hardware/en/latest/pages/BW1099.html) (which means PCBA is opensource), and the other folder is for the actual OAK-D-Lite, which has closed source PCBA (as Intel's VPU is on the PCBA itself), so we can only opensource enclosure/PCBA 3D/step files.
Thanks, Erik

gmacario, 2022-01-16:
Thanks @Erol444 for your super-fast response!

All is clear now, closing issue

closed 2022-01-16 by gmacario