luxonis/oak-hardware #216: Could you add Y-Adapter (KS) Step file to this repo.

bgilreath, 2022-02-24:
The exterior is most necessary to integrate into other designs. Thanks!

Luxonis-Brandon, 2022-02-25:
Good idea.  I thought we had this published but asking internally.

Luxonis-David, 2022-02-25:
@bgilreath you can find dedicated folder for Y-adapter [here](https://github.com/luxonis/depthai-hardware/tree/master/DM6010_Y-adapter).
Step files for the whole product are in [Mechanical folder](https://github.com/luxonis/depthai-hardware/tree/master/DM6010_Y-adapter/Mechanical), in case you would need a step file of a PCBA it is in [3D Models folder](https://github.com/luxonis/depthai-hardware/tree/master/DM6010_Y-adapter/3D_Models).

bgilreath, 2022-02-25:
Thank you! Is it appropriate to also link to it on the OAK-D-Lite page under the 3D models section? https://docs.luxonis.com/projects/hardware/en/latest/pages/DM9095.html

closed 2022-02-25 by bgilreath

Luxonis-David, 2022-02-25:
@Erol444 thoughts on above?