luxonis/oak-hardware #223: Oak-1-W and Oak-1-Global?

justin-larking-pk, 2022-03-11:
Any plans in the future for an Oak-1 variant with a wide angle view lens and/or a global shutter camera to help with motion blur etc? I've been using the oak-1 and oak-1-poe and while they work great they lack the flexibility to do a a lot of tasks, i.e. when I need a higher field of view or global shutter camera to help with situations with there's a lot of movement. Of course this is in specific situations where stereo depth is not needed. else wise i'm aware there the oak-d-w etc.

Luxonis-Brandon, 2022-03-11:
Hi @justin-larking-pk ,

Thanks.  Great suggestion.  Yes, we are planning this.  See below:
[image]

So actually if you would like to order one of these variants, you can do so via our store now (but note this is a customized order so it will take ~3 months) by order OAK-1, and then specifying the customization desired with the customization coupon.  The same options for OAK-D-W (and so on) will be available in OAK-1 via this technique.  

To do so, order both of these below:

1. https://shop.luxonis.com/collections/usb/products/megaai-kit
2. https://shop.luxonis.com/collections/beta-store/products/modification-cupon

And specify which sensor and lens that you would like.  

Arducam short FPC
-----------------

* IMX378

  * NFOV AF
  * NFOV FF
  * WFOV FF (120° DFOV)

* IMX214

  * NFOV AF
  * NFOV FF
  * WFOV (120° DFOV)

* OV9282

  * WFOV FF NoIR (Visible+IR pass) (150° DFOV)
  * WFOV FF IR-block (150° DFOV)
  * NFOV FF NoIR (Visible+IR pass)
  * NFOV FF IR-block

* OV9782

  * WFOV FF IR-block (150° DFOV)
  * NFOV FF IR-block

Luxonis-Brandon, 2022-03-12:
Unfortunately we don't have a PoE modular camera version just yet.  We'd need to take that on as a design project.  

Background: the OAK-1-PoE design still has the old Sunny-style connector for the IMX378, which is particular to only that sensor.  So we'd need to re-do that part of the design and order an additional batch.  We already did this re-do for OAK-1, but not yet OAK-1-PoE.

Feel free to shoot us an email at sales@luxonis.com if it's of interest in contributing to having this made (e.g. if you would like to do a volume order or help cover costs of making this new version).

justin-larking-pk, 2022-03-14:
Ah only looking for two of this particular design at the moment. However does this mean that the CCM connectors would be standardised so that any sensor can be swapped out for another? i.e. imx378 for an ov9782 in the same spot? I had previosuly tried to modify an oak-d to essentially an oak-d-w but ran into the issue that the ov9782 supplied by arducam cant connect in the imx378's place.

Luxonis-Brandon, 2022-04-11:
Shoot sorry @justin-larking-pk I just noticed I missed this.  

Yes, they are being standardized like you mention here.  And yes IMX378 and OV9782 can then be swapped on the PCBA.  With the only change needed between the two being the mechanical portion in front of it (i.e. so that the aluminum doesn't block the wide FOV optics).

And actually you can `jam` a Wide FOV OV9782 (150° DFOV) into this new OAK-1 design (with the internal modular CCM connector), it just looks a bit funny and the glass doesn't fit on (as the optics would hit it):
[image]

So we're making a new front for Wide FOV that makes this fit more naturally:
[image]

stephansturges, 2022-04-24:
... and here is the final version!
[image]

Spoiler alert: they are awesome, and can be configured to output an excellent rectilinear image with high light-sensitivity preserving a fair chunk of the wide FOV!

Luxonis-Brandon, 2022-05-04:
The photos above are with OV9782 (global shutter color 1MP).  And OV9282 (global shutter grayscale).

[image]
[image]
[image]
[image]
[image]

justin-larking-pk, 2022-05-05:
Looks good! Any indication on when these may be available to order? Also any indication on when POE variants with these features will be available?

Luxonis-Brandon, 2022-05-10:
Thanks!  And sorry about the delay.  You can order here:
https://shop.luxonis.com/collections/beta-store/products/oak-1-w

And then leave notes w/ any specificity if you want color, global shutter, etc.  E.g. OV9282 wide FOV, or OV9782 wide FOV, or IMX378 wide FOV.  

And for the PoE.  Good call.  We should do the same there too.  Will ask.

justin-larking-pk, 2022-11-30:
Any update on a wide FOV POE variant?

Luxonis-David, 2022-12-02:
@justin-larking-pk OAK-1-W-PoE is currently in production and we are waiting for first batch to be finished mid December.
After that we will be starting fulfilling backorders, you can expect all variants in [Early Access Store](https://shop.luxonis.com/collections/early-access) soon.
CC: @cafemoloko