luxonis/oak-hardware #248: OAK-FFC-6P - 6-Port OAK FFC

Luxonis-Brandon, 2022-05-18:
### Start with the `why`:

With the new OAK-SoM-MAX ([Series 3](https://docs.luxonis.com/projects/hardware/en/latest/pages/articles/oak-s3.html); [here](https://docs.luxonis.com/projects/hardware/en/latest/pages/DM3399.html#oak-som-max)) it is possible to connect 6x cameras.

For prototyping it is often desirable to have a way to connect various cameras/image sensors/etc.  So an equivalent of our OAK-FFC-3P (which works with OAK-SoM and OAK-SoM-IoT) and OAK-FFC-4P (which works with OAK-SoM-Pro and OAK-SoM-Pro-KB) would be quite useful.

### Move to the `how`:

Leverage the OAK-SoM-MAX module to make an OAK-FFC-6P which supports 6 FFC cameras.

### Move to the `what`:

 - 6x 2-lane MIPI interfaces (note each can still support 4k/60FPS on IMX378 and IMX477/IMX577).
 - uSD Card Slot
 - OAK-SoM-MAX connection
 - IMU
 - USB3.1
 - IO including framesync
 - Coincell battery slot for Linux RTC

closed 2022-05-19 by Luxonis-Brandon

Erol444, 2022-05-20:
Some renders of the board:
[image]
[image]
[image]
[image]

Erol444, 2022-05-23:
Hello @madgrizzle , There should be support to use any pair of the 6 cams to run stereo - we have to ask the HW engineer why he labeled it this way:)

Luxonis-Brandon, 2022-07-04:
[image]
[image]