luxonis/oak-hardware #276: Do you guys have some CE? EMC or RoHS Directive?

OAKChina, 2022-08-03:
[image]

Luxonis-Brandon, 2022-08-03:
Asking team.

Luxonis-David, 2022-08-04:
@OAKChina you mean the paper right?
I have searched through our files and unfortunately we didn't buy any CE directive so far.

RoHS is just Restriction of Hazardous Substances directive, it is sometimes is also referred to "lead-free" directive.
If product is declared as RoHS all components, part of the product must not consist hazardous substances or the concentration level must be below the specified one. 
Something about is is also written [here](https://en.wikipedia.org/wiki/Restriction_of_Hazardous_Substances_Directive).

Luxonis-David, 2022-11-30:
@OAKChina can we close this issue? 
Did you get the answer you were looking for with the above post?