luxonis/oak-hardware #283: What does it take to add an IMU to an OAK-D-Lite PCB?

heyangHEY, 2022-08-08:
I am learning vio recently and need camera+IMU data, but I only have an oak-d-lite on hand. I found that the circuit for the IMU (BNO085) is reserved on the board
[image]
And found the schematic and PCB from OAK_D_LITE_DEV, now there are two problems:
[image]
[image]
1. The PCB design of lite_dev and lite at the IMU are not the same. I understand that the pcb of lite is not open source due to Myriad X, but can you post the component layout of the IMU part of the lite?
2. Do I need to update the firmware of BNO085 and oak-d-lite after soldering on the IMU? I have consulted before and learned that I need to use https://github.com/luxonis/depthai/tree/UI-test-tools-Production to update the firmware of the IMU, but I don’t know the specific method, so could you please give me a general operating route?

Luxonis-David, 2022-08-09:
Hi @heyangHEY,

1. sure we I have attached the schematic part of the IMU on the R2M1E4 version of the board you have. 
2. I normally did the same along with the testing of the device, @Erol444 might know some other options
[DM9095_IMU.pdf](https://github.com/luxonis/depthai-hardware/files/9290082/DM9095_IMU.pdf)

heyangHEY, 2022-08-09:
@Luxonis-David, Thank you for your timely reply. I marked the component numbers in the schematic diagram on the PCB. Do you think I marked it correctly? In addition, what component is the part with the question mark in the lower right corner? Does the lack of it affect the IMU?
[image]
[image]

Luxonis-David, 2022-08-10:
Sorry about the delay @heyangHEY, and I should have made a screen shoot of this section of the PCB in first place (I forgot there is no silk screen there :) ).
So yes you are correct on all, and the component with question mark does not relate to IMU but stereo camera and does not need to be populated. 
I am attaching the screen shot of this area of the PCB which confirms the above. 
[image]

heyangHEY, 2022-08-10:
thx, time to weld the IMU to the PCB. But I'm not clear on the next steps, how can I contact @Erol444, email him?

Erol444, 2022-08-10:
Hi @heyangHEY , no need, I am following this Issue:) I think you can flash the firmware to the IMU using [ImuNode.enableFirmwareUpdate(True)](https://docs.luxonis.com/projects/api/en/latest/components/nodes/imu/#_CPPv4N3dai4node3IMU20enableFirmwareUpdateEb).
Thanks, Erik

ricardodeazambuja, 2022-10-16:
@heyangHEY, did it work at the end???

pulsar666, 2023-02-21:
@heyangHEY I'm curious to know if this worked out for you. I have an OAK D lite too that doesn't have the IMU chip on it and want to add it for my project