luxonis/oak-hardware #292: [Errata] OAK-FFC-CBA - DM0260 R0M0E0

Erol444, 2022-08-19:
We have found that the OAK-FFC-CBA (DM0260) revision R0M0E0 contains a hardware issue which results in camera not being able to run. **Please send an email to support@luxonis.com for the OAK-FFC camera board replacement**.

## Hardware solution

There's an easy hardware fix for this - you need to cut/lift pin 4 of the `U3` voltage regulator: 
[image]

Here you can see how we cut the pin 4, which makes the camera work as expected:
[image]

[image]

Luxonis-David, 2022-08-19:
TL;DR on error explanation:

DM0260_R0M0E0 Printed Circuit Boards were produced with an error on gerbers, resulting in 2V8 regulator not working. The issue manifested only in the latest batch by using alternative part for LDK120C28R. 

The issue is in fourth pin on U3 part on PCBA being connected to GND instead left floating. We didn't detect it on first control batch using TLV70028DCK part, as the operation of the later mentioned regulator is not interrupted by this error. 
Updated design files R1M0E0 were added to GitHub which could help to ease location the issue PR is [here](https://github.com/luxonis/depthai-hardware/pull/293). 
All future boards will be manufactured with based on the R1M0E0 design files removing the issue.

Luxonis-David, 2022-08-19:
Merged to master, project folder location [DM0260_OAK-FFC-CBA](https://github.com/luxonis/depthai-hardware/tree/master/DM0260_OAK-FFC-CBA).