luxonis/oak-hardware #308: cm4 not booting after increasing gpu memory to max

ShakhzodbekYuldoshov, 2022-10-21:
I have a 1 cm4 device. I increased GPU memory from setting to max 820 or something I cannot remember. It asked to reboot. After rebooting the device's display became rainbow colored square display. The device is not turning on. What should I do to make it work?

Erol444, 2022-10-23:
Hi @ShakhzodbekYuldoshov ,
I have never tried it, but you might want to flash a new image to the device, [tutorial here](https://docs.luxonis.com/projects/hardware/en/latest/pages/guides/raspberrypi.html). Let us know if that resolves the problem.
Thanks, Erik

ShakhzodbekYuldoshov, 2022-10-28:
Thank you for your answer. Yes I wanted to flash new image. But when I put sd card to device it was using old image. Then I found this forum discussion and it really helped to me to solve that issue: https://discuss.luxonis.com/d/612-how-to-reset-the-oak-cm4/7
It is combination of tutorial that you have and youtube video about "How to flash the eMMC on a Raspberry Pi Compute Module 4"