luxonis/oak-hardware #324: OAK-SOM-PRO-KB not functioning in OAK-1-POE or OAK-D-POE

robotwhispering, 2022-11-29:
Today I installed a brand new OAK-SOM-PRO-KB into an OAK-1-POE and an OAK-D-POE. The indicator lights on the OAK board light up, and I can see green inside of the Ethernet adapter. But it doesn't really blink, so there doesn't appear to be any  traffic.  The device manager doesn't  recognize the device when this SOM is installedntalled wit hin.

Luxonis-David, 2022-11-29:
Hi @robotwhispering,

Which revision of the OAK-1/D-PoE you are using?
(It is printed on the silkscreen on the PCB.)

Luxonis-David, 2022-11-29:
@robotwhispering I don't get you response. 
Anyway based on the revision of the board we can define what would be the modification needed in order that we can get the OAK-SoM-Pro-S3 working. 
I must tell you that if the boards are the latest revision we wouldn't be able to get the ETH in working state because of the SoC errata on the PCIe clock reference generator. 
Let me know, 
Regards,
David

robotwhispering, 2022-11-29:
R2M1E2 OAK-D-POE
R2M1E1 OAK-1-POE

I thought it showed in the photos

⁣Warm regards, 

Scott Uneberg 
918.883.8842​

Luxonis-David, 2022-11-30:
Got it, so in case of the OAK-D-PoE:
- change GPIO BOOT mode on the DIP slide switch to 0X0F for eMMC boot used on OAK-SoM-Pro-S3.  (now being set to 0x03)
- you would also need to remove the resistor connecting PCIe_ISOLATE to the bootstrap pin, 
- first time using the OAK-SoM-Pro-S3 you will also need to download the image on it, @Erol444 can you assist on that 
- (second time flashing the eMMC you need to use recovery slide to put device in USB recovery mode) 
- I am not sure though if we supported ETH bridge on PCIe already (using external reference clock) CC: @themarpe 

[image]

WRT OAK-1-PoE:
Same mods needed, we can check details after you successfully bring up OAK-D-PoE.

themarpe, 2022-11-30:
ETH bridge isn't supported on RVC3 - if that will change down the line is TBD.
We'll prepare the necessary parts for you to get up and running