luxonis/oak-hardware #326: OAK TrappedRobot

Luxonis-David, 2022-12-08:
Start with the `why`:

Luxonis is being active on the market for more than 3 years now, observing and learning from the market, we are seeing there’s a bifurcation:

1. Robots that are intuitively a _robot_:
   1. Things that drive, walk, crawl, swim, fly, grab things (arms), etc. 
2. Robots that aren’t intuitively a _robot_.
   1. What we called `trapped robots` as they have the _exact_ perception needs as all the the robots that walk, crawl, swim, fly, and grab things (arms)
   2. But these robots _only perceive_
   3.  They don’t actually move, they’re like R2D2 on the back of an X-wing.  Just looking out, and maybe sending signals for other things to do things (like shoot, in R2D2’s case).  

So as we have observed this in the market, there seem to be 3 popular types of `trapped robots` so far:

1. Production lines
   1. In factories for QA/QC automation
   2. The Series 1 **OAK-D-PoE** and Series 2 **OAK-D-Pro-PoE** seem to fit this bill quite well. We are extending this lineup with new product that supports M12 lenses adding more options on optics side with so called **OAK-D-LR**  
   3. And specifically, this market is filled with engineers, engineers who expect M12 X-Coded PoE, etc.
   4. We are likely already covering this market for `trapped robots`
2. Retail.
   1. We do not have a (standard) product that fits this market intuitively as of now.  
   2. Luxonis thesis here is that the following is needed:
      1. Omni view for people tracking.  To make it easier to hand off between one camera to another (hence Luxonis prior work in OAK-D-W models - which similarly helps in autonomous mobile robots).
      2. Detailed view for getting requisite information of retail interactions (what was looked at, who is who w/ accurate re-ID, product details, and requisite details to identify shoplifters/bad-actors).
3. Logistics.
   1. In warehouses, particularly the non-super-huge and non-super-sophisticated (i.e. the majority of the market), where there are no engineering teams managing everything (unlike production QA).  
   2. Relatively simple tasks like bar code scanning, QR-code scanning, box-size measurement and people counting are needed.  And these are “constrained enough” problems that we can very likely make a camera for them.  
4. Transportation
   1. We saw many companies that wanted to build smart spatial dash cams based on our platform  
   2. This covers all sorts of mapping applications, including the mapping in huge sites/fields in agriculture not necessarily only in transportation 

So whereas Luxonis already has options for 1 (Production Lines) above which are products targeting hardware engineers, we do not cover 2 and 3 yet.  Customers are telling us that there is a need for 2 and 3 and that they’d buy them, and that competing products are stocking out, it may make sense for Luxonis to make cameras for each.  

Move to the `what`:

Make a new series of “**OAK TrappedRobot**”. Idea on the name being “**TrappedRobot**” is that it should stand out among a sea of “Yet Another Security Camera” and get across the point that this is indeed different, should also get enough attention so that it would be easy to remember.  

It also conveys the idea that this has capabilities that are for automation… not just for security.  It’s a “bot” doing something.  

The two products in this series would/could be to address Retail and Logistics above.

The first one we started with is so called **TBot** which would for a bit follow the products we are already familiar with, but on the other hand it offers plethora of additional options that someone might not be used to, working with to date introduced, Luxonis OAK products. 

**TrappedRobot (TBot)**
[image]

It will cover logistics and light manufacturing.  It will also be used with **TBot-360°** in retail that will be another product in the series.  The idea is to have an easy-to-setup and “just works” camera that non-engineers can use for things such as:

- Box measurement
- QR and/or bar-code scanning 
- People counting/trac
- Basic quality/inspection/etc.

Initial specifications we are looking into are: 
- 2x 48MP rolling shutter 120° DFOV camera 
  - 1x IR capable 
  - 1x IR filtered 
- 3x 1MP global shutter stereo camera FOV (80° HFOV)
  - 7.5cm H-stereo
  - 3.0cm V-stereo
- 1x IR laser dot projector
- IR LED/s 
- uSD card slot --> covered with a silicon rubber, on the bottom side - supporting outdoor installs
- USB --> accessible and covered with a silicon rubber
- PoE on the back with vertical RJ45 connector 
- RVC3 based product (Linux host)
- Large system indication LED
- Factory reset button, next to the ethernet connector
- Camera mount included (wall mounting, pole mounting)
  Indoor/Outdoor installation
- Microphones - 2 in front and 1 on the back side
  
[image]
  

As mentioned there will be a series of  **OAK TrappedRobot** products, after **OAK-TBot** we are looking after introducing the following:

- **TrappedRobot 360°** **(TBot-360°)**
- **TrappedRobot Transit (TBot-Go)**
- **TrappedRobot Multi (TBot-Multi)**

Specifications and details about the products will follow along with the development phase and prototyping phase.

Luxonis-David, 2022-12-08:
[image]

Erol444, 2024-02-02:
Due to the OAK4 prioritization (based on the new RVC4 chip), we have decided against pursuing the Sense lineup.

closed 2024-02-02 by Erol444