luxonis/oak-hardware #329: OAK-FFC-6P UART w/ SOM-MAX

steveturner, 2023-01-04:
The AUX headers on the OAK-FFC-6P have breakouts for DBG_UART_TX and DBG_UART_RX as shown in [the production schematic](https://github.com/luxonis/depthai-hardware/blob/master/DM3390_OAK-FFC-6P/Docs/Schematic%20PDF/Production.PDF),
[image]

Are these mapped to a serial port within the Yocto image or firmware somehow? We've done an initial test and they do not seem to map to /dev/ttyS[0-3]

Are the other UARTS accessible such as the one currently attached to the IMU?

Erol444, 2023-01-04:
cc @themarpe on this

themarpe, 2023-01-07:
@steveturner 

These are 1v8 and map to UART3 block which is internally used as console. (It unfortunatelly cannot be used for non-console things)

According to the current pinmux, there aren't any other UARTs muxed out.

IMU uses SPI.

SPI_SIO2 could be used as UART1_RX, and SPI_SIO1 as TX but that is muxed for IMU SPI also

[image]

steveturner, 2023-03-14:
These pins are indeed mapped and in the default firmware and OS show up as `/dev/ttyS3`. The port is in use for debugging purposes during initial boot and for login to the OAK-SOM-MAX once booted into linux.

closed 2023-03-14 by steveturner