luxonis/oak-hardware #335: CM4-PoE driver install instructions

jacobzweig, 2023-01-19:
We had to swap a RPi module from an OAK-D CM4 to a n OAK-D-CM4-PoE and now we're unable to interact with the cameras, with error messages like:
```bash
Traceback (most recent call last):
  File "/home/pi/luxonis/utils/encoding_example.py", line 28, in <module>
    with dai.Device(pipeline) as device:
RuntimeError: No available devices
```

Additionally `lsusb | grep 03e7` does not return anything.

Erik pointed us here to open an issue for instructions to install the drivers for the USB hub on the CM4-PoE and hopefully recover this device. 

Thanks!

Erol444, 2023-01-21:
Hi @jacobzweig ,
Sorry about the delay. At [RPI docs](https://docs.google.com/document/d/1c4JhIY2ulcgaaycUBOsAbUI1EJ_7Zx4ux4alaxFc_ts/edit) see `General-purpose RaspberryPi OS image` V2 image prep:
```
- Clone rpi-eeprom to ~/Documents/rpi-eeprom, checkout to `45935788fddd94fb2c17105855500d345d751187`. 
- Uploaded [FW013705.bin](https://luxonis.slack.com/archives/C023SAH5X09/p1628063682050400) to ~/Documents/rpi-eeprom/firmware/waveshare/FW013705.bin so it can be flashed with
sudo ~/Documents/rpi-eeprom/firmware/vl805 -w ~/Documents/rpi-eeprom/firmware/waveshare/FW013705.bin (reboot afterwards)
```
Could you try that?
Thanks, Erik

Erol444, 2023-01-21:
And of course you would need the .bin:)
[FW013705.zip](https://github.com/luxonis/depthai-hardware/files/10472682/FW013705.zip)

jacobzweig, 2023-01-24:
Thanks Erik,

Tried this and got 
```bash
jacob@oak0:~/Documents/rpi-eeprom $ git checkout 45935788fddd94fb2c17105855500d345d751187
HEAD is now at 4593578 github: Disable blank issues

jacob@oak0:~/Documents/rpi-eeprom $ sudo ~/Documents/rpi-eeprom/firmware/vl805 -w ~/Documents/rpi-eeprom/firmware/waveshare/FW013705.bin
* no VL805 found
```

Rebooted, but we're still getting the same error

```
Setting up demo...
  File "/home/pi/depthai/depthai_demo.py", line 655, in run
    self.instance.run_all(self.conf)
  File "/home/pi/depthai/depthai_demo.py", line 131, in run_all
    self.setup(conf)
  File "/home/pi/depthai/depthai_demo.py", line 179, in setup
    self._deviceInfo = getDeviceInfo(self._conf.args.deviceId, args.debug)
  File "/home/pi/.local/lib/python3.9/site-packages/depthai_sdk/utils.py", line 222, in getDeviceInfo
    raise RuntimeError("No DepthAI device found!")
RuntimeError: No DepthAI device found!
```

any other ideas?

Erol444, 2023-01-26:
cc @Luxonis-David on this. Otherwise @jacobzweig perhaps it would be best if you reflashed the image with the newest one.
Thanks, Erik

Luxonis-David, 2023-01-26:
@jacobzweig I suppose that you got the new OAK-D-CM4-PoE devices which does not anymore integrate VL805 HUB we had issues with. So VL805 won't be detected and for the same reason you don't need to flash the EEPROM.
New latest image would be needed here or adding drivers for UPD720201K8-701-BAC-A as we do OTP the driver from CM4.
Does that makes sense? Hopefully this solves the issue.    
Thanks!

jacobzweig, 2023-01-26:
Thanks @Erol444 and @Luxonis-David – is it possible to install the drivers for UPD720201K8-701-BAC-A without re-flashing the device? I no longer have physical access to it.

themarpe, 2023-01-26:
@jacobzweig it is - what do you see when running just `lsusb`?

Erol444, 2023-01-26:
@jacobzweig could you try steps below?

```
# Host
scp out.tar.gz pi@[ip]:~
# On RPi
cd ~ && mkdir tmp
tar -xf ~/out.tar.gz

sudo cp etc/systemd/system/rvc2-prepare.service /etc/systemd/system/cm4-rvc2-usb-prepare.service
sudo mkdir -p /lib/firmware/renesas/
sudo cp lib/firmware/renesas/K2026.mem /lib/firmware/renesas/
sudo cp usr/sbin/rvc2-upd72020x-check-and-init /usr/sbin/
sudo cp usr/sbin/upd72020x-load /usr/sbin/

cd ~ ; rm -r out.tar.gz ; rm -r tmp 

sudo systemctl enable cm4-rvc2-usb-prepare
```

[out.tar.gz](https://github.com/luxonis/depthai-hardware/files/10513573/out.tar.gz)

jacobzweig, 2023-01-26:
Thanks @Erol444 – this worked!

closed 2023-01-26 by jacobzweig