luxonis/oak-hardware #346: OAK PoE connection to moving robot

Franko9999, 2023-02-08:
Hi guys,

Today I received my OAK PoE. Didnt know the connection comes through the ethernet cable. Was thinking it uses Wifi or something. I need to connect it to the moving robot (moving distance can be up to 50 m). Connecting robot to ethernet cable is not an option. The target would be to move the robot around from a distance with laptop or mobile phone for example and see the computer vision model inspections right on device. Maybe some of you have experience on how to achieve overall setup. It would be nice if you could help me in this.

Best regards,
Roberts

stephansturges, 2023-02-08:
You need one of these:
https://linitx.com/category/poe-powerbank-/1142/149,1142

Then connect over wifi. I do that all the time and it works great.

stephansturges, 2023-02-08:
Yep that works. Another way is to use the powerbank to inject power into an ethernet cable using a splitter, and connect the data side to your robot.

Franko9999, 2023-02-08:
Hi Stehpan,

The second option would mean the data would flow directly to the robot. I
would connect to the the robot through wifi. What kind of setup would I
need to receive the camera data from robot? Would this setup be faster in
terms of data transfer compared to the previous one I mentioned? Im new in
robotics. It would be nice if you could help to understand this.

Best regards,
Roberts

stephansturges, 2023-02-08:
I'm afraid I don't understand the question. If your robot can connect over ethernet to the camera then this setup will work... but if you don't know how the robot works already then I can't say whether you can connect a camera 🙃

Franko9999, 2023-02-08:
Do you know some alternatives besides these ones: https://linitx.com/category/poe-powerbank-/1142/149,1142

Shipping costs for these to EU (Latvia) are approximately the same as the price for the product.

Best regards,
Roberts