luxonis/oak-hardware #356: OAK SoM Pro - is USB2.0 necessary?

0y8w1x, 2023-02-20:
hey, I'm designing a board with the OAK SoM Pro and saw that the SoM breaks out USB_D_N and USB_D_P, pin 90 and pin92 respectively. Is that lane necessary to route to the host, or can the device operate normally without that USB2.0 lane?

[](url)
[image]

Luxonis-David, 2023-02-21:
@0y8w1x USB2 can be used as side channel for debugging and it is highly recommended to route it to the SoM-Pro at least on the new designs unless you are certain that you don't need it. I don't know for sure if USB2 is mandatory in case you would need to flash the NoR with pipeline/bootloader, but can be that you can use only SS pairs for that as well. 
CC: @Erol444 on USB flashing over USB3

themarpe, 2023-02-21:
@Luxonis-David USB flashing over USB3 only (before BL, etc...) is not possible. Internal BL supports USB2.

Luxonis-David, 2023-02-21:
Thanks @themarpe, @0y8w1x routing USB2 is necessary.

0y8w1x, 2023-02-21:
thanks. I'd like to ask then, is it possible to use the USB2.0 lane on a seperate USB connector? lets say to only flash the bootloader over the USB2.0 connector, and use the device after that with only the SS lanes?

Luxonis-David, 2023-02-21:
Yes that is possible, not sure what would be the reason though?

0y8w1x, 2023-02-21:
here is what's going on, the main IC in my design has 1x type-c input (alt mode pin assignment D) and 1x USB3.0 input. it then muxes the ML and USB3.0 together on one type-c output. with 'type-c' input I mean 2x SS pairs and 1x USB2.0 plus cc and so on. and then I only have one USB3.0 lane left. the manufacturer suggests a USB2.0 hub, but I don't want to do this if there is an easier alternative. and it sounds like I don't need it for normal operation, just once to flash the bootloader. and on top of that the routing is nicer ^^

Luxonis-David, 2023-02-22:
I don't think I follow the above architecture description, if you would need any assistance would you be able to provide a simplified block diagram explaining the architecture you are mentioning?

0y8w1x, 2023-02-22:
[image]
let me know if this explains it better

Luxonis-David, 2023-02-22:
Yes using HUB would be best in this case if you need to go through ASIC or need ASIC connected to USB2.

closed 2023-02-22 by 0y8w1x