luxonis/oak-hardware #361: Sensor HW issue on my OAK-D-LR?

chengguizi, 2023-03-01:
Hi Luxonis Team,

I might be getting a faulty sensor on `camc` of my OAK-D-LR

[image]

Just to show how it supposed to look like, this is how the `camb` look like under the same condition.

[image]

I have tried in different lighting conditions, different exposure and iso. The vertical green bands persist. May I know if this is confirmed to be a hw quality issue? I dont have this issue on another OAK-D-LR unit I bought together.

themarpe, 2023-03-17:
Cc: @Luxonis-David

Luxonis-David, 2023-03-17:
Hi @chengguizi,

We've noticed this phenomena on another unit and is related to sensor issue likely damaged during assembly. It is strange that this slipped our QC though, hence question, were the green lines present on the sensor from a start?  We will have to check if we can replace your device right away or we will have to wait on next batch.
Will get back to you an sorry for any inconvenience!

chengguizi, 2023-03-17:
@Luxonis-David  It was there right at the start. So very likely a miss in the QC stage.

Erol444, 2023-03-17:
Hi @chengguizi , Then it would be best to send an email to support@luxonis.com and we will get it covered. Also cc @njezersek about QC.

Erol444, 2024-02-02:
HW issue, the device has been replaced, closing due to inactivity.

closed 2024-02-02 by Erol444