luxonis/oak-hardware #429: non functioning camera port on RVC3 6P-FFC module

PeterQFR, 2023-07-31:
I'm having an issue where one of my cameras on a 6P-FFC module, particularly the Left camera socket is not getting data from the attached camera. The module seems to recognise there is a camera there, but when I add the camera using the depthai-python/utilities/cam_test.py it fails to produce images and blocks other cameras from retrieving their images. 

This is the output with DEPTHAI_DEBUG=1

```
$ python cam_test.py -cams left,m right,m rgb,m camd,m came,m -cres 800 -fps 4
[2023-07-13 17:37:44.206] [info] DEPTHAI_DEBUG enabled, lowered DEPTHAI_LEVEL to 'debug'
[2023-07-13 17:37:44.206] [debug] Python bindings - version: 2.19.1.0.dev+11a40ed426e44610d40f1a328eb6a054d9af313f from  build: 2023-07-10 16:43:48 +0000
[2023-07-13 17:37:44.206] [debug] Library information - version: 2.19.1, commit: 0ee1ebf76bfd8181b701dfcf781fb6402b26f0ff from 2023-07-10 16:08:00 +0000, build: 2023-07-10 16:28:42 +0000
[2023-07-13 17:37:44.208] [debug] Initialize - finished
Enabled cameras:
   left : mono
  right : mono
    rgb : mono
   camd : mono
   came : mono
DepthAI version: 2.19.1.0.dev+11a40ed426e44610d40f1a328eb6a054d9af313f
DepthAI path: /home/peter-laptop/depthai-python/venv/lib/python3.8/site-packages/depthai.cpython-38-x86_64-linux-gnu.so
left
right
rgb
camd
came
[2023-07-13 17:37:44.300] [debug] Resources - Archive 'depthai-bootloader-fwp-0.0.22.tar.xz' open: 2ms, archive read: 90ms
[2023-07-13 17:37:44.675] [debug] Resources - Archive 'depthai-device-fwp-b511dc1a23aa9016007f7d6e9979a5a322bcefbe.tar.xz' open: 2ms, archive read: 465ms
[2023-07-13 17:37:44.724] [debug] Device - OpenVINO version: 2022.1
[2023-07-13 17:37:44.724] [debug] Device - BoardConfig: {"camera":[],"emmc":null,"gpio":[],"logDevicePrints":true,"logPath":null,"logSizeMax":null,"logVerbosity":null,"network":{"mtu":0,"xlinkTcpNoDelay":true},"pcieInternalClock":null,"sysctl":[],"uart":[],"usb":{"flashBootedPid":63037,"flashBootedVid":999,"maxSpeed":4,"pid":63035,"vid":999},"usb3PhyInternalClock":null,"watchdogInitialDelayMs":null,"watchdogTimeoutMs":null} 
libnop:
0000: b9 0f b9 05 81 e7 03 81 3b f6 81 e7 03 81 3d f6 04 b9 02 00 01 ba 00 be be bb 00 bb 00 be be be
0020: be be be 01 bb 00
[2023-07-13 17:37:44.735] [debug] OS version: , Gate version: 
[2023-07-13 17:37:44.735] [debug] DeviceGate createSession: {"fwp_version":"0.0.1+57109fe11bf03b4cb130b9afcc6a13246692b825","library_version":"2.19.1","name":"depthai_session","protected":true}
[2023-07-13 17:37:44.746] [debug] DeviceGate createSession response: {"connection_information":"","create_datetime":"2022-07-29T17:36:34.664436","destroy_datetime":null,"fwp_checksum":null,"fwp_exists":false,"fwp_version":"0.0.1+57109fe11bf03b4cb130b9afcc6a13246692b825","id":"13","library_version":"2.19.1","name":"depthai_session","owner":"","protected":true,"start_datetime":null,"state":"CREATED","stop_datetime":null}
[2023-07-13 17:37:45.665] [debug] DeviceGate upload fwp successful
[2023-07-13 17:37:46.178] [debug] DeviceGate start fwp successful
Connected cameras: [{socket: RGB/CENTER/CAM_A, sensorName: OV9282, width: -1, height: -1, orientation: AUTO, supportedTypes: [], hasAutofocus: 0, name: keembay_cam_a2}, {socket: LEFT/CAM_B, sensorName: OV9282, width: -1, height: -1, orientation: AUTO, supportedTypes: [], hasAutofocus: 0, name: keembay_camex_0_stereo}, {socket: RIGHT/CAM_C, sensorName: OV9282, width: -1, height: -1, orientation: AUTO, supportedTypes: [], hasAutofocus: 0, name: keembay_camex_1_stereo}, {socket: VERTICAL/CAM_D, sensorName: OV9282, width: -1, height: -1, orientation: AUTO, supportedTypes: [], hasAutofocus: 0, name: keembay_cam_c1}, {socket: CAM_E, sensorName: OV9282, width: -1, height: -1, orientation: AUTO, supportedTypes: [], hasAutofocus: 0, name: keembay_cam_c2}]
 -socket RGB   : OV9282   -1 x   -1 focus:fixed - 
 -socket LEFT  : OV9282   -1 x   -1 focus:fixed - 
 -socket RIGHT : OV9282   -1 x   -1 focus:fixed - 
 -socket CAM_D : OV9282   -1 x   -1 focus:fixed - 
 -socket CAM_E : OV9282   -1 x   -1 focus:fixed - 
USB speed: UNKNOWN
IR drivers: []
[2023-07-13 17:37:48.173] [debug] Schema dump: {"connections":[{"node1Id":10,"node1Output":"out","node1OutputGroup":"","node2Id":9,"node2Input":"in","node2InputGroup":""},{"node1Id":8,"node1Output":"out","node1OutputGroup":"","node2Id":7,"node2Input":"in","node2InputGroup":""},{"node1Id":6,"node1Output":"out","node1OutputGroup":"","node2Id":5,"node2Input":"in","node2InputGroup":""},{"node1Id":4,"node1Output":"out","node1OutputGroup":"","node2Id":3,"node2Input":"in","node2InputGroup":""},{"node1Id":2,"node1Output":"out","node1OutputGroup":"","node2Id":1,"node2Input":"in","node2InputGroup":""},{"node1Id":0,"node1Output":"out","node1OutputGroup":"","node2Id":10,"node2Input":"inputControl","node2InputGroup":""},{"node1Id":0,"node1Output":"out","node1OutputGroup":"","node2Id":8,"node2Input":"inputControl","node2InputGroup":""},{"node1Id":0,"node1Output":"out","node1OutputGroup":"","node2Id":6,"node2Input":"inputControl","node2InputGroup":""},{"node1Id":0,"node1Output":"out","node1OutputGroup":"","node2Id":4,"node2Input":"inputControl","node2InputGroup":""},{"node1Id":0,"node1Output":"out","node1OutputGroup":"","node2Id":2,"node2Input":"inputControl","node2InputGroup":""}],"globalProperties":{"calibData":null,"cameraTuningBlobSize":null,"cameraTuningBlobUri":"","leonCssFrequencyHz":700000000.0,"leonMssFrequencyHz":700000000.0,"pipelineName":null,"pipelineVersion":null,"xlinkChunkSize":-1},"nodes":[[10,{"alias":"","id":10,"ioInfo":[[["","out"],{"blocking":false,"group":"","name":"out","queueSize":8,"type":0,"waitForMessage":false}],[["","raw"],{"blocking":false,"group":"","name":"raw","queueSize":8,"type":0,"waitForMessage":false}],[["","frameEvent"],{"blocking":false,"group":"","name":"frameEvent","queueSize":8,"type":0,"waitForMessage":false}],[["","inputControl"],{"blocking":true,"group":"","name":"inputControl","queueSize":8,"type":3,"waitForMessage":false}]],"name":"MonoCamera","parentId":-1,"properties":{"boardSocket":4,"cameraName":"","fps":4.0,"imageOrientation":-1,"initialControl":{"aeLockMode":false,"aeRegion":{"height":0,"priority":0,"width":0,"x":0,"y":0},"afRegion":{"height":0,"priority":0,"width":0,"x":0,"y":0},"antiBandingMode":0,"autoFocusMode":3,"awbLockMode":false,"awbMode":0,"brightness":0,"chromaDenoise":0,"cmdMask":0,"contrast":0,"effectMode":0,"expCompensation":0,"expManual":{"exposureTimeUs":0,"frameDurationUs":0,"sensitivityIso":0},"frameSyncMode":0,"lensPosAutoInfinity":0,"lensPosAutoMacro":0,"lensPosition":0,"lowPowerNumFramesBurst":0,"lowPowerNumFramesDiscard":0,"lumaDenoise":0,"miscControls":[],"saturation":0,"sceneMode":0,"sharpness":0,"strobeConfig":{"activeLevel":0,"enable":0,"gpioNumber":0},"strobeTimings":{"durationUs":0,"exposureBeginOffsetUs":0,"exposureEndOffsetUs":0},"wbColorTemp":0},"numFramesPool":3,"numFramesPoolRaw":3,"resolution":1}}],[9,{"alias":"","id":9,"ioInfo":[[["","in"],{"blocking":true,"group":"","name":"in","queueSize":8,"type":3,"waitForMessage":true}]],"name":"XLinkOut","parentId":-1,"properties":{"maxFpsLimit":-1.0,"metadataOnly":false,"streamName":"came"}}],[8,{"alias":"","id":8,"ioInfo":[[["","out"],{"blocking":false,"group":"","name":"out","queueSize":8,"type":0,"waitForMessage":false}],[["","raw"],{"blocking":false,"group":"","name":"raw","queueSize":8,"type":0,"waitForMessage":false}],[["","frameEvent"],{"blocking":false,"group":"","name":"frameEvent","queueSize":8,"type":0,"waitForMessage":false}],[["","inputControl"],{"blocking":true,"group":"","name":"inputControl","queueSize":8,"type":3,"waitForMessage":false}]],"name":"MonoCamera","parentId":-1,"properties":{"boardSocket":3,"cameraName":"","fps":4.0,"imageOrientation":-1,"initialControl":{"aeLockMode":false,"aeRegion":{"height":0,"priority":0,"width":0,"x":0,"y":0},"afRegion":{"height":0,"priority":0,"width":0,"x":0,"y":0},"antiBandingMode":0,"autoFocusMode":3,"awbLockMode":false,"awbMode":0,"brightness":0,"chromaDenoise":0,"cmdMask":0,"contrast":0,"effectMode":0,"expCompensation":0,"expManual":{"exposureTimeUs":0,"frameDurationUs":0,"sensitivityIso":0},"frameSyncMode":0,"lensPosAutoInfinity":0,"lensPosAutoMacro":0,"lensPosition":0,"lowPowerNumFramesBurst":0,"lowPowerNumFramesDiscard":0,"lumaDenoise":0,"miscControls":[],"saturation":0,"sceneMode":0,"sharpness":0,"strobeConfig":{"activeLevel":0,"enable":0,"gpioNumber":0},"strobeTimings":{"durationUs":0,"exposureBeginOffsetUs":0,"exposureEndOffsetUs":0},"wbColorTemp":0},"numFramesPool":3,"numFramesPoolRaw":3,"resolution" [... cut ...]
When I exclude this port the system works reliably up to 4 cameras. I did have issues when attaching to port camf that would prevent other cameras from working. 

I've rotated the cameras around and have confirmed that all the cameras do work and I've reattached a camera to the left socket a couple of times and am confident that that connection is a correct as I can make it. 

The previous discussion is [here](https://discuss.luxonis.com/d/1841-oak-ffc-6p-left-socket-not-working/3)

I have 5 cameras so let me know if there is some test you would like me to do.

Erol444, 2023-08-01:
Hi @PeterQFR ,
Which version of the FIP image is flashed to the board?

PeterQFR, 2023-08-01:
Hi What is FIP and how do I find out. I am on the rvc3_develop branch of the depthai-python  11a40ed426e44610d40f1a328eb6a054d9af313f

PeterQFR, 2023-08-06:
Hi @Erol444  Erik gave me the references to the FIP and I ensured that 1.5.0 was installed. It hasn't changed anything specificially. I have cameras on RIGHT, RGB, CAMD, CAME, CAMF. Individually they all produce images however when I run them all together they are detected but don't produce any frames.