luxonis/oak-hardware #439: OAK-D to 4-Screw Adapter

knwachuk, 2023-08-28:
## Start with the `why`:

[image]

I am working on a project and I need the OAK-D camera to interface with the 4-screw (3 in the picture) Raspberry Pi Camera Holder of the [WAVESHARE AI Jetson Nano Kit](https://www.waveshare.com/product/robotics/mobile-robots/jetbot-ai-kit-acce.htm).

## Move to the `how`:

I need an adapter to go from the mounting solution of the OAK-D to the 4-screw version and I suspect that 3D printing is my solution.

## Move to the `what`:

I am really looking for either of two things:

1. Is there an already existing solution to this problem in the form of an STL file?
2. Is there a more elegant solution?

Erol444, 2023-08-29:
Hi @knwachuk ,
We have some T-slot adapter STL files, perhaps you could use those as the starting point? They'd use the vesa mounting screw holes in the OAK-D.
https://github.com/luxonis/depthai-hardware/tree/master/adapters

Erol444, 2024-12-19:
Closing due to inactivity

closed 2024-12-19 by Erol444