luxonis/oak-hardware #483: Upstream of BW1099 to Kicad repositories

phodina, 2024-06-15:
Hi,

just want to check you're okay with upstreaming the BW1099 module into the official Kicad repositories. At the end you list MIT so that shouldn't be a problem :-)

I've modified the schematics and split it into 3 units (power, connectivity and cameras).

Also generated the wrl model to be used for the 3D rendered board view.

It's not yet ready as there as some minor changes I still want to do and check it against KLC (CI hooks to check for conformity).

Here are the links for the commits if you want to check in the meantime:

- Symbol:
https://gitlab.com/kicad/libraries/kicad-symbols/-/commit/63ddd383fdd6c2e1c0ae546d3ff25cddcf370b3d

- Footprint:
https://gitlab.com/kicad/libraries/kicad-footprints/-/commit/23a18c0331b3d4baebe10db8eb4f39777671378e

- 3D Package:
https://gitlab.com/phodina/kicad-packages3D/-/commit/549337cf03c8599ad66699d5c0d7ab164ef6b3e3

Erol444, 2024-06-15:
Hi @phodina ,
Of course, feel free to do so! And thanks for sharing & refactoring/cleaning the project:)! We might also update our schematics to yours, our HW team will check it once the `luxonis_bw1099` is merged.
Thanks again, Erik

phodina, 2024-06-18:
FIY @Erol444 just waiting for someone to review it

-3D package:

https://gitlab.com/kicad/libraries/kicad-packages3D/-/merge_requests/1048

- Symbol:

https://gitlab.com/kicad/libraries/kicad-symbols/-/merge_requests/4590

- Footprint:

https://gitlab.com/kicad/libraries/kicad-footprints/-/merge_requests/3700

phodina, 2024-07-14:
@Erol444 I've fixed the offset issue in the WRL model and the source model is also submitted as Freecad project.

I'm not Mechanical engineer but I noticed that the model is missing mounting holes. Are they missing in the STEP file or is that due to some import/conversion bug?

[image]

Erol444, 2024-12-19:
@phodina can we close this off:)?

closed 2025-04-18 by phodina