luxonis/oak-hardware #485: OAK-FFC 4P PoE IR

oscarthorn, 2024-07-02:
Hi!
What's the correct way to connect OAK-FFC IR to OAK-FFC 4P PoE? The IR module has a 10 pin cable is there an adapter for the camera connectors?

Erol444, 2024-07-02:
Hi @oscarthorn , OAK-FFC-IR only works with FFC-4P ([mentioned in shop description](https://shop.luxonis.com/products/oak-ffc-ir)), the FFC-4P-POE, unfortunately, does not have the 10pin connector. We apologize for the inconvenience.

oscarthorn, 2024-07-03:
Ah, I see! And there is no way to drive it of one of the ffc connectors for the cameras? They seem to have the needed lanes and I can get the 5V from somewhere else since they only seem to be 3.3V. But I suppose the software support is lacking? The problem for me is that I want both the Ethernet and IR, both I suppose I could evaluate one at a time and then get both later.

Erol444, 2024-07-03:
@oscarthorn it's possible, if you're willing to hack around with tiny wires. I'd suggest checking FFC-IR and FFC-4P schematics to determine how to connect it. Software support shouldn't be lacking, you'll just need to connect i2c so RVC2 chip can control the controller on the FFC-IR (intensity of projector/led).

oscarthorn, 2024-07-03:
Great! I'll just have a cable made, no problem.

closed 2024-07-03 by oscarthorn